AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #1: B&W Menu Bar Icon

ajkblue, 2020-04-17:
Would it be possible to add an option to the app to make the menu bar icon black and white?

davidwernhart, 2020-04-17:
Hi @ajkblue !

Thank you for your feature suggestion.
Currently, I am very busy working on my bachelor's thesis, so I sadly don't have much time for this project right now.
I  will be able to continue in 1 - 2 months, so stay tuned.

Best regards,
David

frowin, 2020-06-03:
> Would it be possible to add an option to the app to make the menu bar icon black and white?

I would also appreciate it a lot! 

@davidwernhart I thought Apple will introduce this feature with the latest update but they didn't. So I'm glad that you built this helper.

closed 2020-06-04 by davidwernhart