AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #2: Thanks for releasing this but I have Installed helper and restarted machine and is not working on 16" Pro - Catalina 10.15.2

soundbanana, 2020-04-17:


davidwernhart, 2020-04-17:
Hi @soundbanana !

Is the application starting and the battery does not stop charging or does the application not even start at all?

If it does start, please try the following:
Drain your battery to about 60%, then set max charging percentage to 70% and start charging your macbook again. Charging should stop at about 73% (It should say "battery is not charging" when you click on the battery icon in the status bar).
I have a 16" macbook with catalina myself and it works for me.

Best regards,
David

soundbanana, 2020-04-17:
Yep I guessed it needed to be below the percentage you choose. I let it drain to 70 and set it the limit above that and now it's working. 

Delighted with it, I've been looking for this feature for years. I saw apple announced that they are adding some form of battery health feature in  a future update but I doubt it will allow manual limits to be set like this. Thanks a bunch, I'd be happy to donate for your work!

davidwernhart, 2020-04-17:
Glad it worked out for you!
In my testing, it should also stop charging when you are above the threshold, although MacOS usually takes pretty long (about 5 minutes for me) to realize and display this.

I also was pretty frustrated finding that there were no tools for this available, especially since the batteries in MacBooks are pretty hard to replace. I am curious what future MacOS versions bring though.

Feel free to donate using the the button located in AlDente's settings, I would very much appreciate it! 

Take care,
David

closed 2020-04-17 by davidwernhart