AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #3: FAQ

dmus123, 2020-05-12:
Easy Yes/No Questions for an FAQ :) 

- If you delete the software, does the SMC go back to default without changing any settings?
- If you reset the SMC, does the SMC go back to default without changing any settings? 
- If I reset the SMC & delete AlDente, does the SMC go back to default without changing any settings?
- Is there anything more to deleting your software other than moving to trash?

Thank you for creating this amazing tool!

davidwernhart, 2020-05-13:
Hi @bobd992, thank's for your interest in my tool!

> If you delete the software, does the SMC go back to default without changing any settings?

No, it does not.That being said, I noticed in my testing, that MacOS sometimes resets the BCLM key when rebooting, especially after a software update.

> If you reset the SMC, does the SMC go back to default without changing any settings?

Yes.

> If I reset the SMC & delete AlDente, does the SMC go back to default without changing any settings?

Yes.

> Is there anything more to deleting your software other than moving to trash?

The only thing AlDente changes is one Key in the SMC (BCLM). Like I said before, MacOS seems to reset this one automatically on some occasions.  If you want to delete AlDente, simply set the slider back to 100% and then move the application to trash.

To change SMC keys (which requires root privileges), AlDente installs a privileged helper tool upon first launch. This tool is not uninstalled when the app is removed. As far as I know, there is no "nice" way to uninstall such a tool, and it is not intended to do so by Apple. Although it is not deleted, it uses virtually no disk space and no system resources. If you want to get rid of unused helper tools, take a look at: [https://forums.developer.apple.com/thread/66821](https://forums.developer.apple.com/thread/66821)

I hope this was helpful!

Best regards,
David

bobd992, 2020-05-13:
Super helpful! I read that forum thread - the idea of a helper app with root access that I can't remove is very scary to me from a security perspective. I wish there were some straightforward safe way to remove it if I wanted to in the future or to avoid having that. Again, software appears to be beautiful and you really solved a huge problem here so kudos!

closed 2020-05-13 by bobd992