AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #5: Launch at Login does not function

tommiyau, 2020-05-23:
I've installed and run this and it works well except the launch at login function does not run. If I look in the system logs it returns an error every 10 seconds that it can not execute the program or that it does not exist. This is on Catalina 10.15.1. Executing the application manually it works fine. Needing to disable the Launch at Login to get rid of the every 10 second attempt to launch.

tommiyau, 2020-05-23:
May 24 09:17:30 MacBook-Pro com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[15968]): Could not find and/or execute program specified by service: 45: Operation not supported: com.davidwernhart.AlDente-LaunchAtLoginHelper
May 24 09:17:30 MacBook-Pro com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[15968]): Service exited with abnormal code: 78
May 24 09:17:30 MacBook-Pro com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper): Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
May 24 09:17:40 MacBook-Pro com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[15972]): Could not find and/or execute program specified by service: 45: Operation not supported: com.davidwernhart.AlDente-LaunchAtLoginHelper
May 24 09:17:40 MacBook-Pro com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[15972]): Service exited with abnormal code: 78
May 24 09:17:40 MacBook-Pro com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper): Service only ran for 0 seconds. Pushing respawn out by 10 seconds.

davidwernhart, 2020-05-26:
Hi @tommiyau!
Thank you for this bug report. I am aware that launch at login is not working right now, I will try and fix it soon when I have more spare time.

Best regards,
David

davidwernhart, 2020-06-04:
This issue should be fixed now in [Release 1.2](https://github.com/davidwernhart/AlDente/releases)

Best regards,
David

closed 2020-06-04 by davidwernhart