AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #7: Only works with Catalina?

cyclism6486, 2020-05-27:
when I open the DMG on my laptops running Mojave I see the dreaded "circle with line" over the app's icon. I really wanted this to work for Mojave since I don't plan on going to Catalina until next September.  Maybe update the readme to state "requires macOS 10.15.x"

krackers, 2020-05-27:
You can compile command line version of `SMCUtil` from https://github.com/sicreative/BatteryStatusShow and run `sudo SMCUtil -w BCLM <hex val>`. Works as far back as 10.9 (probably even further, as it just needs the smc key to be defined).

alexwhittemore, 2020-05-27:
If you want a slightly easier way, I built a quick wrapper around that same utility in the form of an Alfred workflow that lets you set charging level to either 70% or 100% cap. 

It's here, midway down below the Alfred screenshot
https://www.alexwhittemore.com/controlling-macbook-pro-charging-cap-with-alfred/

godly-devotion, 2020-05-28:
> You can compile command line version of `SMCUtil` from https://github.com/sicreative/BatteryStatusShow and run `sudo SMCUtil -w BCLM <hex val>`. Works as far back as 10.9 (probably even further, as it just needs the smc key to be defined).

@krackers Do you have a sample build command for the `SMCUtil`? I'm assuming you would need to build the `smcutil/main.m` file.
Also, what would be the hex value format, say for example for 80%?

alexwhittemore, 2020-05-28:
@krackers you can also take the easy route and just download it pre-compiled, which is what I did for that Alfred workflow :) https://github.com/sicreative/BatteryStatusShow/tree/master/BatteryStatusShow/smcutil/Products/usr/local/bin

cyclism6486, 2020-05-28:
@krackers - just because I figured out how to leave an "issue" here doesn't mean I'm a developer.  Last time I used a compiler was with Fortran 77, so I have no idea how to do what you said. @alexwhittemore - I will take a look at that Alfred workflow....but first I need to install Alfred.

godly-devotion, 2020-05-28:
Thanks for the tips everybody. I decided to create a small utility using Script Editor to set the max charge level using `smcutil`. It should run on almost all versions of macOS, though it was specifically tested on my MacBook Pro 13" (early 2015) with macOS Mojave 10.14.6.

https://github.com/godly-devotion/charge-limiter

krackers, 2020-05-29:
@alexwhittemore Nice work! I believe the pre-compiled binary uses the `os_unfair_lock` symbol which was only defined in 10.12+.

https://github.com/sicreative/BatteryStatusShow/blob/4e64c47a8d6c51416ad9fd250650929b327162c8/BatteryStatusShow/smc.c#L140

 To build for older platforms it should be safe to replace this with the commented out variant (`OSSpinLock`)

davidwernhart, 2020-06-04:
Hi guys!

Since I developed and compiled this tool on Catalina, I was not able to test if it works for other versions of MacOS. I am sad to hear that it apparently does not.

Also @alexwhittemore congrats on your Alfred workflow, it seems to be very handy and convenient!

Best regards,
David

closed 2020-06-04 by davidwernhart