AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #9: Does it work with Catalina 15.5?

dtoniolo, 2020-05-29:
macOS Catalina 15.5 introduces the new Battery Health feature. However, it doesn't substitute the functionality of AlDente, which I'd like to keep.

Do you have tested it with 15.5?

maiksd, 2020-05-29:
I'm testing right now, with a MBP 15" 2018. At first it seemed like it didn't work at all. I had to start AlDente and set a target _after_ disabling and re-enabling 15.5 battery health management, and now charging has stopped. But more testing is needed, this is inconclusive.

nunodl, 2020-05-30:
Working fine here on a 2019 MBP 13" (catalina 10.15.5) with crappy battery health feature turned OFF. Just had to give permission in security and privacy settings.

maiksd, 2020-05-30:
Which permission were that exactly? I can't find any an entry for AlDente there.

nunodl, 2020-05-30:
I've got the "cannot be opened because the developer cannot be verified."
Had to do this
https://support.apple.com/en-us/HT202491

dtoniolo, 2020-06-01:
Just installed Catalina 15.5.5. Worked straight away without having to disable Battery Health Management. Keep in mind that I don't have AlDante lauched at startup.

CAUTION: as soon as I launched alDente, it reset itself to 100% (real, not nominal, charge, i.e. >100% of what you normally see). If you leave it like that you might damage the battery.

More testing needed, but it seems to work.

Wiskatcha, 2020-06-02:
I personally can't make it work, on my "MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports)", under macOS 10.15.5.
I tried with and without the new "Battery health" option.
I'll try to see if I can compile and debug a bit... ?

Anyway, thanks @davidwernhart for this tools. I'd dream of this kind of feature and you brought it :)

schniko, 2020-06-03:
It's not working with my 2020 MBP either :-( I'm looking forward for a features like this for years. This software is so great!

dtoniolo, 2020-06-04:
Yep, nice suggestions. I'd like to code them myself, but I don't now anything about SMC and macOS

davidwernhart, 2020-06-04:
Hey guys!
Thank you for your nice words and feedback!

For me, AlDente works without any problems on 15.5 both with the battery health feature enabled and disabled. That being said, my MacBook is fairly new and I don't know if anything changes over time (Since it seems that the battery health feature does not do anything right now for me).

@Wiskatcha thank you for your suggestions!

> start alDente with an intermediary value of 50 or 70%
Since the maximum value is saved in the SMC anyway, it should be sufficient to set your desired value at first start. It will usually keep this value when rebooting.

> forbid the cursor to go above the value macOS would stop
AlDente already does this! The cursor only works on values between 20% and 100% (which is the default value your MacBook ships with)

> turn a color to red, or add a "danger" sign when above this value
I consider taking this in account for future releases, although there is not really a "dangerous" setting. As described above, until OSX 15.5, all MacBooks were at 100% at all time, which is not the most healthy thing but also not dangerous.

Also, I changed the metrics with the new [release](https://github.com/davidwernhart/AlDente/releases) to match the percent values that MacOS shows.

I hope this helped!

Best regards,
David

JanJastrow, 2020-06-08:
@davidwernhart 
Hi, thanks for the tool - it sounds really interesing.
Sadly I downloaded it and could not run it - it requires Catalina (which I'm still skipping).
Could you add a note to the readme that this tools requires 10.15?

davidwernhart, 2020-06-09:
Hi @JanJastrow 

Thank you for your feedback!
I added the note to the readme.

Best regards,
David

closed 2020-06-09 by davidwernhart