AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #15: 1.2 Release File Damaged

GreenRaccoon23, 2020-06-04:
After installing version 1.2 from the [releases](https://github.com/davidwernhart/AlDente/releases) page, I am unable to open the app. I receive this message:

> “AlDente.app” is damaged and can’t be opened. You should move it to the Trash.

I did the following:
1. Downloaded the .dmg.
2. Opened the .dmg.
3. Dragged the app to the Applications directory.
4. Opened "AlDente" with Launchpad.
5. Saw the error window.

I also tried launching the app directly from the Applications directory with the same result.

I am running Catalina 10.15.4 on a MacBook Pro A1989 (2018).

davidwernhart, 2020-06-04:
Hi @GreenRaccoon23!

Is this still the case with the new dmg in the release?
I updated the archive a few minutes ago because of other reports of this problem.

Sorry for the inconvenience,
David

peterlewis, 2020-06-04:
Howdy. It's still an issue for me also, with the latest `AlDente.1.2.dmg` with SHA256 `18564d8a6db70b01a69fbdfe7935e4d4fee70dd0c934f2c266d16c1ed8f77e89`.

davidwernhart, 2020-06-04:
Thank you for your feedback!

I am sure that this is an issue with the .dmg archive rather than the app.
Sadly, I cannot really test this since everything works just fine for me.

I just uploaded a new signed version of the dmg file at the 1.2 release.
Please let me know if this works for you!

Best regards,
David

peterlewis, 2020-06-04:
Unfortunately this one(`AlDente.1.2_signed.dmg` with SHA256 `8d05dc61a352320d2a29c218970e9141c9c5079f524a78dadb77af5d50211a40` won't even let me open the DMG. I'm now getting:

`“AlDente.1.2_signed.dmg” cannot be opened because it is from an unidentified developer.`

Sorry to be the bearer of bad news...

davidwernhart, 2020-06-04:
This is really weird.
Is it possible to open the dmg using the old "settings-trick" ?
https://www.macworld.co.uk/how-to/mac-software/mac-app-unidentified-developer-3669596/

peterlewis, 2020-06-04:
Ah I've spotted the issue, I think. Looks like you're signing it using your Apple Development cert!

davidwernhart, 2020-06-04:
Yes I do, but is there any other way to do it?

Does it work for you now?

peterlewis, 2020-06-04:
I believe you'll need to sign this using `Developer ID Application`: See random example, below:

```
Executable=LittleSnitch-4.5.2.dmg
Identifier=LittleSnitch-4
Format=disk image
CodeDirectory v=20200 size=302 flags=0x0(none) hashes=1+6 location=embedded
Hash type=sha256 size=32
CandidateCDHash sha256=0b0f7f6d397d580f10ed9533fff9e4ee1ba559e9
CandidateCDHashFull sha256=0b0f7f6d397d580f10ed9533fff9e4ee1ba559e9ac254a444438f40372f94b50
Hash choices=sha256
CMSDigest=0b0f7f6d397d580f10ed9533fff9e4ee1ba559e9ac254a444438f40372f94b50
CMSDigestType=2
Page size=none
CDHash=0b0f7f6d397d580f10ed9533fff9e4ee1ba559e9
Signature size=9006
Authority=Developer ID Application: Objective Development Software GmbH (MLZF7K7B5R)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=25 May 2020 at 11:21:46
Info.plist=not bound
TeamIdentifier=MLZF7K7B5R
Sealed Resources=none
Internal requirements count=1 size=176
```

davidwernhart, 2020-06-04:
I don't think that I can do that since I am not enrolled in the Apple Developer Program. 99$ a year would be a big loss for a project like this. The only signing identity available to me is: Apple Development: David Wernhart (HXBYV744D5).

I have now created yet another release with signing certificate "Sign to Run Locally" instead of "Developement".

Maybe this will work now...
Anyway, thanks for your help!

davidwernhart, 2020-06-04:
in addition to that, I now also uploaded a zip archive of the app. Hopefully, this does the job.

peterlewis, 2020-06-04:
Ah right, I'm sorry, I didn't realise! I believe (but I may be wrong) that, if you don't have a signing cert, then the best thing would be to leave it completely unsigned and allow people to bypass as required. Please do someone correct me though, if I'm wrong.

peterlewis, 2020-06-04:
Unfortunately the .zip is still reporting to be damaged. It appears that it's still signed, as follows:

```
Executable=/Applications/AlDente.app/Contents/MacOS/AlDente
Identifier=com.davidwernhart.AlDente
Format=app bundle with Mach-O thin (x86_64)
CodeDirectory v=20500 size=1573 flags=0x10000(runtime) hashes=40+5 location=embedded
Hash type=sha256 size=32
CandidateCDHash sha256=607ea3f7e1c9c4f77f5039d2fd48bdef858ceb72
CandidateCDHashFull sha256=607ea3f7e1c9c4f77f5039d2fd48bdef858ceb7235ccbf128f29a049d1d38dcd
Hash choices=sha256
CMSDigest=607ea3f7e1c9c4f77f5039d2fd48bdef858ceb7235ccbf128f29a049d1d38dcd
CMSDigestType=2
Page size=4096
CDHash=607ea3f7e1c9c4f77f5039d2fd48bdef858ceb72
Signature size=4752
Authority=Apple Development: EMAIL-REDACTED (GSDX9BQ584)
Authority=Apple Worldwide Developer Relations Certification Authority
Authority=Apple Root CA
Signed Time=4 Jun 2020 at 13:55:37
Info.plist entries=27
TeamIdentifier=56C2L92EKW
Runtime Version=10.15.4
Sealed Resources=none
Internal requirements count=1 size=204
```

I speculate that the fact that the .app is signed with an Apple Development cert is the likely cause, but I may be wrong there!

davidwernhart, 2020-06-04:
Yes, I also don't understand why it does not work now while it worked for the first version.
A friend of mine did not get the initial .dmg file to work, but the zipped app worked without any problems. Like I already said, on my mac it works every time, making this a very hard problem to debug. I recompiled the project one last time and uploaded the zip archive. It probably won't work again but it is worth a try.

I will look into this problem more in-depth in the course of the next week.
Thank you for your patience,
David

peterlewis, 2020-06-04:
Ah okay, so I've just redownloaded the .zip and I was able to bypass Gatekeeper and open AlDente 1.2 without any issues! I apologise, it may be that I forgot to bypass Gatekeeper when opening the .app from the .zip, before! :-/ Unfortunately I've since deleted the other .zip so I can't re-verify.

No need to thank for patience! Thank YOU for making this app! As someone who uses my MacBook Pro plugged-in at my desk, all day, every day, It's something I've been hankering after for EONS! :)

All the best,

Peter

ajkblue, 2020-06-04:
I can also vouch for the latest ZIP release working!
I had issues with prior releases today as well, but now it installed and works perfectly.

GreenRaccoon23, 2020-06-05:
The latest zip works for me too.

When I opened it, I got the message `"AlDente.app" can’t be opened because Apple can’t check it for malicious software.` I had to override the security settings temporarily in order to open it. A [link](https://www.macworld.co.uk/how-to/mac-software/mac-app-unidentified-developer-3669596/) in an [earlier comment](https://github.com/davidwernhart/AlDente/issues/15#issuecomment-639129173) shows how to do this. The official Apple documentation for it is [here](https://support.apple.com/en-us/HT202491) under "How to open an app that hasn’t been notarized or is from an unidentified developer".

Thank you everyone for your help and thank you `davidwernhart` for the app!

davidwernhart, 2020-06-09:
Hi guys!

Sorry for the late response and thank you for your immediate feedback!

I am very happy that this last version turned out working. I still don't know why the dmg archives (I tried two different methods of generating them) have been corrupting the app.
Since it works, I will simply stick with the .zip version and update the install instructions in the readme for now.

Best regards,
David

closed 2020-06-09 by davidwernhart

shooter366, 2020-11-19:
> After installing version 1.2 from the [releases](https://github.com/davidwernhart/AlDente/releases) page, I am unable to open the app. I receive this message:
>
>
> I did the following:
>
> 1. Downloaded [...]

Open settings > Security> confirm opening.