AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #16: macOS battery scale inaccurate at lower percentages

peterlewis, 2020-06-04:
Howdy, no rush on a fix, but just wanted to highlight (and note down, before I forget) where AlDente appears to get 'macOS battery scale' percentages slightly wrong, lower down the charging scale. All measurements done on 2020 MacBook Pro 13" (A2257).

AlDente 53% = Real 51%

I'll update this as I find more. Maybe a pattern can then be established! :)

davidwernhart, 2020-06-09:
Hi @peterlewis,

thank you for your involvement and feedback! (And your generous donation!)

You are right, the new macOS battery scale feature is still not 100% exact for all percentages. In fact, enabling the checkbox simply removes 3% from the set value and sets the SMC with that, which is admittedly quite a hack.
It appears that the conversion between the real charging values and the ones displayed by macOS is not that simple. The gap seems to get bigger the higher the value is, but I have not figured out the pattern yet. Also, I don't know if this conversion difference is the same for all MacBooks or changes when the battery ages.

Anyway, I will keep my eye on this.

Best regards,
David

peterlewis, 2020-09-05:
@davidwernhart Just to re-visit this. Just over two MacBooks (16-inch 2019 and 13-inch 2020) I've noticed discrepancies, both between the two, as well as on each individual machine when setting maximum at varying percentages.

Also, applications such as 'Coconut Battery' report a more accurate battery percentage anyway, so even if it were to be consistent across different percentages/models, it doesn't feel like it's adding benefit.

Please can I suggest that the 'Use macOS battery scale' is either removed or at least that the setting is honoured across re-launch?

I hope that doesn't sound like a rant. I adore this application and keen to see grow! 👍 

Many thanks!

Peter

closed 2021-07-05 by MatthiasKerbl