AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #19: How to deinstall?

burnout-syndrome, 2020-06-08:
Thank you for your great work, I love the idea of this project! But how can I get back to the old state, in case that I want to deinstall the app? Especially because of the helper, how can I remove it completely?

davidwernhart, 2020-06-09:
Hi @burnout-syndrome!

Thank you for your feedback!
As I previously described in [other](https://github.com/davidwernhart/AlDente/issues/4)issues, the helper tool is not a real concern regarding security, performance or battery life and this system is also used by many other applications.
If you want to uninstall the application, simply set the maximum charge percentage back to 100 and move the app to trash.
If you want to remove the helper tool, you first have to locate the location of the helper.
This can be accomplished with a script like [this](https://github.com/erikberglund/Scripts/blob/master/tools/privilegedHelperToolStatus/privilegedHelperToolStatus).
Then, you have to make the script executable (`sudo chmod 755 privilegedHelperToolStatus.sh`) and execute it with `sudo ./privilegedHelperToolStatus.sh`. This will show you all installed helpers and wether or not the application the helper is built for still exists on your system. For me, the location of AlDente's helper tool is `/Library/PrivilegedHelperTools/com.davidwernhart.Helper`. Navigate to the folder and delete the file.

I hope that this helped.

Best regards,
David

closed 2020-06-09 by davidwernhart

burnout-syndrome, 2020-06-10:
Great, thank you for your reply!

halpz, 2021-02-15:
AppCleaner finds all these files too!

Kerndog73, 2021-05-10:
The homebrew formula also properly deletes this file

BANKslow, 2021-06-03:
I am a leek and am not able to run these commands properly. I am stuck on putting terminal to the right folder where the HelperStatusTool script is located. Would you be so kind to ELI5 how to delete the Helper application through terminal?

sufyazi, 2021-06-15:
> I am a leek and am not able to run these commands properly. I am stuck on putting terminal to the right folder where the HelperStatusTool script is located. Would you be so kind to ELI5 how to delete the Helper application through terminal?

Hi @BANKslow to remove a file from the terminal, just run 'rm [the file you want to delete]'. rm is a terminal command called 'remove', and the space is important so as the right name of the thing you want to delete. You said you were able to point the terminal to the right folder where the script you want to delete is located, so just run `rm [name of file]` with no brackets and keeping the single space between rm and the filename. Press enter after the line and it'll be deleted.