AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #22: Not working with MacBook 12' (2017)

crzydg, 2020-06-12:
HI there,

I was waiting for this feature for ages, and now, that it seems to be available, it doesn't work on my MacBook:
MacBook (Retina, 12-inch, 2017)
10.15.5 (19F101)

It charges up and above 100% although I installed (and re-installed) the helper several times.

Sad :-(
-crzydg

burnout-syndrome, 2020-06-13:
Same here... MacBook Air 2014, running macOS 10.15.5. Instead of stop charging at 80%, my battery health has decreased from 94% health to 87...

klabalaba, 2020-06-15:
First I thought, it doesn't work on my MacBook (12", 2017). I installed the 1.0 Version and tried again. Then after a while it worked. I noticed it after the battery charging was below 80%. Then I plugged in the charger and it stopped charging at 80%.

I am not sure what exactly did the trick, but I think it was the 1.0 Version and some waiting...

burnout-syndrome, 2020-06-15:
How long did you wait? I reinstalled version 1.0, waited an hour and nothing happened ...

crzydg, 2020-06-15:
The 1.0 version didn't work for me at all (in all settings, restarts, reinstalls etc.) so I'm just now testing 1.2. - will give feedback if that does the trick.

klabalaba, 2020-06-15:
For me just the 1.0 version works, on MB 12" as well as MB Pro 16". On the 16" MBP it worked after a few minutes (1.2. didn't work at all). On the 12" MB I can't say for sure... I only noticed it the next day.

burnout-syndrome, 2020-06-15:
I have now tested both versions, none of them work for me.

burnout-syndrome, 2020-06-20:
For whatever reason, it suddenly works. Maybe it's because i previously hid the symbol with bartender. I only noticed that the Magsafe LED remains orange when the set battery level is reached and does not switch to green. And the question remains why my battery health went down so badly.

klabalaba, 2020-07-21:
Update: 1.2 works for me as well. On three MacBooks now (12" MB 2017, 13" MBP 2014, 16" MBP 2019). 

About the thing with battery health: I think it didn't really went down and it's just a calibration issue. Since (afaik) the battery calibration is dependend on a full charge. So maybe if you use your battery and then charge it to 100% the health is going to increase again. I noticed it on my MBP 16". After some more charging the health went from 94-95% back to 97%.

andrewjoe, 2020-08-01:
I can confirm 1.2 works for me on Macbook 12 (2016).

Diamonstar, 2020-12-15:
Confirming works on Macbook 12 (early 2015) 1.2Ghz 8,1 running OS Big Sur.  Limiting charge to 60% has regained battery health from mid 80's to 90% now.  Only charge to 100% when planning business trips now.      Many thanks to developer!

closed 2021-02-20 by TheMazus