AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #23: Battery health dropped since using AlDente

burnout-syndrome, 2020-06-14:
Since the app ran on my MacBook Air 2014, which by the way did not work, my battery health has dropped from 94 to 85 percent. Checked with coconut battery. Deleting the app and an smc reset did not bring any improvement. What happened here?

davidwernhart, 2020-06-23:
Hi @burnout-syndrome!

Sadly, I cannot tell you why your battery health dropped that significantly. Like I already stated in the readme, AlDente is only modifying a key in the Mac's SMC and I cannot guarantee anything since I do not have the resources to test this app on every MacBook. I do know for a fact, that the measured battery health does fluctuate a lot because of measurement inaccuracies and this may be amplified since your MacBook already has a few years under the belt. It seems like AlDente is working without any problems for the vast majority of users. I am really sorry if it does not for you! If You still have aforementioned problems, I will add your MacBook to the list of unsupported devices.

Best regards,
David

burnout-syndrome, 2020-06-23:
Well, it has happened now and cannot be changed. I was only interested in why battery health dropped so suddenly. Otherwise, the app works for me now, after it initially didn't work. Can it be set up somehow that the Magsafe LED switches to green when the set value is reached? Otherwise I would not put the computer on the list of unsupported devices, after all it works now.

ApricumMedia, 2020-08-04:
Hey, just out of curiosity, @davidwernhart — I'm assuming you originally wrote this utility for yourself to use? Are you still actively using it, maybe even across different machines and did you notice an increased lifetime (or a slower climbing cycle count)?

In burnout's case, if the app didn't even work in limiting charge as they said, I can not imagine a single run causing any physical changes at all to the battery, though it did bring me to think if the effects are measurably positive even if they should logically be beneficial.

davidwernhart, 2020-08-05:
Hi @ApricumMedia!
I am indeed still actively using it. Although I was able to test the general functionality on a few devices of my colleagues, the only device I have been long term testing AlDente is my 16" MacBook Pro purchased in January 2020. So far, I have experienced no negative side effects. I currently have 115 cycles and my battery health is fluctuating between 96 and 100%. Of course I don't know how my battery health would be without AlDente but it does not seem to have had a negative impact so far. From the theory that this app is based on, real benefits should start be measurable when using a MacBook and the tool for many years.

It is most certain that the app will not work on EVERY mac, and I am sadly not able to go to the bottom of every case, hence the disclaimer. AlDente now has 3500+ downloads with a comparatively small number of complains, from which I am concluding that it is working for most users.

Hope that helps,
David

diogoterremoto, 2020-09-26:
I think battery health readings differ with the battery charge. Charge it to 100% of its capacity and then check if health has changed.

dlintin, 2020-10-07:
> Since the app ran on my MacBook Air 2014, which by the way did not work, my battery health has dropped from 94 to 85 percent. Checked with coconut battery. Deleting the app and an smc reset did not bring any improvement. What happened here?

Im having the same issues, battery drop from 87% to 81% in one week usage, but the strange things is, after deleting the app, few minutes later the battery indicator back to 87%! 

im doing a test with a voltage/amp meter, to see if the app if its really working, and the result is, it's actually working, when the limit set to 70%, volt/amp meter showing almost 0 usage!, so im installing this app back, maybe the percentage issues just a bug because using this app, but not harming a battery at all.

dlintin, 2020-10-07:
> Since the app ran on my MacBook Air 2014, which by the way did not work, my battery health has dropped from 94 to 85 percent. Checked with coconut battery. Deleting the app and an smc reset did not bring any improvement. What happened here?

have you tried to uninstal the app, and wait for few minutes? it seems to fix the problem, and fix the wrong indicator of the battery capacity, after that you can install the app again, if its work for you, maybe the percentage issues just a bug because using this app, but not harming a battery at all.

koltony, 2021-03-10:
I dont think that its harming the battery in any way especially this much, the only way this could have happened if you would overcharge it, but if i remember correctly you cant really do that.
Battery health works by measuring the inward and the outward currents and compare that to the battery's designed capacity. For this to be accurate you need to have at least one charge cycle. The more battery you drain before charging, and the more percentage you charge back the more accurate the health indicator will be. Since your battery with this app cant discharge much the reading will be very inaccurate, thats why you see this.
Just turn off the app, use it like a normal laptop for a few days and the health indicator will go back to normal.

ZSH-HSZ, 2021-03-17:
same problem

kmhwang, 2021-04-09:
I'm using the app and set it to 50% from the very beginning of battery replacement. The charge is slowly going up in clamshell mode with dock connected. When I was using the Mac, can see not charging which I thought the app is working.

After a few days, it reach 100% finally, but at 87.8% health .
[image]

I've done 100-0-100-0-100, the health seems not coming back.

Will test two more cycles, and report back.

calindotgabriel, 2021-06-12:
I also saw a quick decline on my m1 mac while using aldente and charger plugged in

dalisoft, 2021-07-18:
After removing app, try these methods, may work for you
- `sudo nvram -c`
- Turn off Macbook
- Close lid
- Wait 30-60 seconds
- Open lid again

If Macbook gets turn on itself (on my M1 Macbook it did itself) and Wi-Fi connection becomes slower than usual, it worked.
My Macbook battery health got back from 99.5% to 100% (has been 99.9% before app installation)