AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #24: Charged past limit

exiva, 2020-06-19:
I installed AlDente on my MacBook Pro (Retina, 13-inch, Early 2015) Running 10.15.5, Set the max to 80%, and discharged the battery to 50%. Plugged it in, and it charged back up to 100% despite the 80% limit. Do you need to reboot or something after the change is set? Is there a way to verify the new limit got set with a command or something?

Love the idea of the app, especially since the native Apple one is disabled on machines that would actually benefit from it. Hopefully can get this working. Thanks!

DevNulPavel, 2020-06-19:
Has charge limit value been saved after your macbook reboot?

zackelia, 2020-06-20:
@exiva I have the exact same MacBook as you, so it should work. I couldn't find an easy way to read the max battery charge level from the terminal so I put together some swift code that you can quickly compile and run to verify. It uses the same SMC.swift from the helper tool appended with a few lines to read the BCLM value. I posted the code in a gist [here](https://gist.github.com/zackelia/c10222c9a79dc3228a1bd9cf067d4079).

```
$ swiftc read-bclm.swift
$ ./read-bclm 
77
```

exiva, 2020-06-20:
@zackelia Hmm. That came back as 100, and any time I change the value in aldente it stays at 100. 🤔 Let me try a reboot and see what happens.

DevNulPavel, 2020-06-20:
https://github.com/sicreative/BatteryStatusShow

Maybe this application will work and change BCLM value.

exiva, 2020-06-20:
Ok, I rebooted and it reset back to 100 in aldente. Now changing it in aldente is being reflected in the check app @zackelia posted. I guess a reboot is required before it is able to modify the SMC params. Going to discharge my battery some and see what happens.

exiva, 2020-06-21:
Rebooting after installing did the trick. Thanks again for the help.

closed 2020-06-21 by exiva