AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #25: Do we need to set the battery percentage on AlDente every time the machine reboot?

randylim27, 2020-06-20:
I realized I always set mine at 80% but when I reboot the machine, it goes back to 103%. Is this normal behavior?

DevNulPavel, 2020-06-20:
@randylim27 Maybe you need to disable default 10.15.5 battery care management system

randylim27, 2020-06-21:
Thanks for the note! Will try that out

randylim27, 2020-06-22:
Hey, i tried that and i've disabled the Mac OS's battery management and I did a reboot today, AlDente still reverts back to 103%.

DevNulPavel, 2020-06-22:
@randylim27 Can you try (BatteryStatusShow)[https://github.com/sicreative/BatteryStatusShow] or my simple script (osx_battery_charge_limit)[https://github.com/DevNulPavel/osx_battery_charge_limit]? Do you have the same behaviour on your computer with this apps?

davidwernhart, 2020-06-23:
Hi guys!

Generally, it is not necessary to set the percentage after every reboot, since the SMC stores this value even if the system is powered off. In my experience, system updates do usually reset the SMC values and you do have to change the max charging percentage afterwards.
It is possible that the new battery management feature messes with the values, although I have not experienced this with my MacBook. I would disable it just to make sure.

Best regards,
David

DevNulPavel, 2020-06-23:
@randylim27 Maybe do you have any other application, which rewrites SMC parameters of your macbook?

un1k0n, 2020-06-30:
@davidwernhart i have read what you wrote but i have two question related to this. Which is the purpose of running aldente at login if the battery level value is not commonly reset by the system it self? and Why aldente shows 103 when reboot and not the last percentage i set before?
Thanks in advance and nice job with this tool.

bmansvk, 2020-08-05:
I have the same question as@un1k0n

> @davidwernhart i have read what you wrote but i have two question related to this. Which is the purpose of running aldente at login if the battery level value is not commonly reset by the system it self? and Why aldente shows 103 when reboot and not the last percentage i set before?
> Thanks in advance and nice job with this tool.
> 
> Used in a 2020 macbook pro 4 thunderbolt ports.

If something rewrites SMC battery charge parameter to 100% (or 103% as shown in AlDente on Macbook Air 2019), AlDente should revert it back to specific value (if running in the background). That behavior seems logical for me.

davidwernhart, 2020-08-06:
Hey @un1k0n , @bmansvk!
Sorry for the delayed answer, I have been quite busy for the last weeks with not much time to spare for this project.

You are right, there is no point of AlDente running in the background besides the convenience of being able to change the maximum charge really fast at any time. Same with the launch on login feature, you might as well leave it off.

The ability to detect external overwrites and apply the set charge level on startup is on my list of upcoming features. I am hoping to find enough spare time and energy over the next weeks to implement this along with some other features/fixes.

Take care,
David

un1k0n, 2020-08-08:
Dont worry, i have been using AlDente since i wrote that comment and i didnt find any issue. Thanks for this great tool.

samleemc, 2020-08-26:
@davidwernhart Take your time! No rush! and I think you can even sell this software at some point to make it up for all the blood and tears you spent over your spare time and weekend!

BigMcGuire, 2020-10-17:
Thank you for this tool! Loving it :) -- So convenient to change without having to type in the admin password all the time.

closed 2021-07-05 by MatthiasKerbl