AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #27: New 2020 13" MBP with 4 TB Ports keeps restarting every 7-10 days since AlDente

invizible00, 2020-06-26:
Hi, first of all, thanks for creating and providing AlDene.

Nonetheless, since I have installed AlDente on my 2020 13" MBP with 4 TB Ports, my Mac had to restart in the morning after opening the lid. Previous shutdown cause is -5 according to the log file. This already happened three times.

The MBP is plugged to a LG monitor, reason why I was hoping AlDente 1.2 would take care of the battery health.

I have to add tho, that the battery management function of 10.5.5 was enabled. I did disable it now, just to check if that might be the reason for the shutdowns and restarts.

Many thanks for any help or advice!

DevNulPavel, 2020-06-26:
It looks like an old problem with external monitors 

https://www.imore.com/some-16-inch-macbook-pros-are-restarting-when-asleep-after-installing-macos-10154-heres-fix

invizible00, 2020-06-26:
> It looks like an old problem with external monitors
> 
> https://www.imore.com/some-16-inch-macbook-pros-are-restarting-when-asleep-after-installing-macos-10154-heres-fix

Thanks for this super fast reply! Hopes were high but Power Nap was and is disabled while plugged in.

invizible00, 2020-06-26:
Just found this:

https://mrmacintosh.com/10-15-4-update-wake-from-sleep-kernel-panic-in-16-mbpro-2019/

Looks like I have to wait until the release of macOS 11.

Already did a SMC reset. Already uninstalled and reinstalled AlDente. Points below do apply to my current settings:
3. Disable Sleep when on Battery “Put hard disks to sleep when possible”
4. Disable “Enable Power Nap While Plugged into a power adapter” or “Enable Power Nap while on battery power”

BigMcGuire, 2020-10-17:
I have a 2020 MBP 13 too and ... it crashes a lot with my external monitor and EGPU - more than my 2017 MBP did. Really annoying. Glad to see Mac OS 11 will be an improvement. <sigh>.

closed 2021-07-05 by MatthiasKerbl