AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #28: Charging does not stop on MacBook Pro 2015 15" running macOS 10.15.5

michaelgosling, 2020-07-03:
Set it to 50 and it didn't stop. Let it drop to 47%, plugged back in, I'm at 88% and climbing. Maybe the new Catalina adding Battery Health stuff messed with it. Or maybe I'm doing something wrong, I kinda hope thats true because I want this to work bad enough I upgraded from Mojave haha...

ludgerus, 2020-07-05:
Hi! I have the same MacBook and the same OS. It did not work with mine either at first but since today it works. I had restarted AlDente (I think) several times to see if it remembers the set percentage. One time AlDente started up with a percentage of 103% although I hadn't set it to that percentage. After that happened I set the percentage to 80% and now it works! There probably be some kind of check and visualisation if the SMC values could be changed as expected. Good luck!

JayBrown, 2020-07-14:
Same behavior and solution on a 2018 MBP with i9 and T2: quit & relaunch AlDente, and it's working… now stuck at 80% and saying "Not charging" in iStatMenus. FYI: I kept Catalina's new battery health management enabled, and it seems that if you do that, macOS will not try to squeeze in these extra percentage points. Charge value stays at 80%, not going to 83% etc.

roko-p, 2021-01-20:
Same on 2017 MacBook Pro 15" running macOS Catalina (10.15.7) -- just exit and re-open AlDente (didn't need to _"Reinstall Helper"_), and it's working :) 👌 

------------------

By the way, I, too, kept the Catalina's _Battery health management_ enabled, **but** for me the percentage still says 2-3% above what I set (which is 70%) (I have the _"Use macOS battery scale"_ setting disabled)

JayBrown, 2021-01-21:
You're right: this was just momentary. Now I'm at 80%, but sometimes also 81% or 82%. But that's still fine, I guess.

closed 2021-07-05 by MatthiasKerbl