AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #30: Request: 10.13 High Sierra

Donothir, 2020-07-21:
Older computers (around 2011) don't have a way to keep their (failing) battery optimal. If this program was made backward compatible to High Sierra, I (my MBP 2011) would be really happy.

DevNulPavel, 2020-07-21:
@Donothir 
You can try my command line alternative, it probably should work. My script uses the same approach as AlDente
https://github.com/DevNulPavel/osx_battery_charge_limit

al3x-huang, 2020-08-09:
Is there a reason why this app doesn't support 10.12? I don't suppose the SMC BCLM logic is different vs 10.15

YvanQu, 2020-12-08:
if you haven't gotten this figured out yet, heres something a bit more user friendly https://github.com/godly-devotion/charge-limiter

MatthiasKerbl, 2021-07-05:
Hi @DevNulPavel,

due to SwiftUI not being compatible with older macOS versions, AlDente is unfortunately not compatible with High Sierra.

closed 2021-07-05 by MatthiasKerbl