AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #31: What is the de-installation?

crzydg, 2020-07-26:
Hi,

I wanted to charge up to full 100% with AlDente. But the charger doesn't accept the "100" field or ignores it. So I set AlDente to not start on reboot, rebooted the MacBook (w/o AlDente) and put it on charging - with the result, that it also only charges to 83% (had set AlDente to 80% previously).
Now, my question is: how can I completely reset my device to original state without AlDente? Is there a hidden magic setting, that I don't see? 
Bonus-question: how can I remove AlDente and all helper scripts safely?

DevNulPavel, 2020-07-26:
@crzydg 

You can try reset your SMC settings, it resets SMC's BCLM flag too. 
https://support.apple.com/en-us/HT201295

For deinstallation you can use AppCleaner up, it worked for me.
https://freemacsoft.net/appcleaner/

crzydg, 2020-07-28:
Hi @DevNulPavel thanks for the hints. Unfortunately none of them worked for me - ok, the AppCleaner does remove the app of course, but my MacBook still only charges up to 89%.
Is there any magic switch I can set in some script to get that fixed?

pinqinband, 2020-07-28:
Hi @crzydg 

I am sorry for your inconvenience!
Resetting the SMC should definitly do the trick!
You can read further in [this](https://github.com/davidwernhart/AlDente/issues/3) thread where I already answered many questions regarding uninstalling.

Hope that helps,
David

Btw, many thanks to @DevNulPavel for answering these issues and keeping a little weight off my shoulders! :)

davidwernhart, 2020-07-28:
Hi @crzydg

I am sorry for your inconvenience!
Resetting the SMC should definitly do the trick!
You can read some more information in this [issue](https://github.com/davidwernhart/AlDente/issues/19) where I already answered some questions regarding uninstalling.

Hope that helps,
David

Btw, many thanks to @DevNulPavel for answering these issues and keeping a little weight off my shoulders! :)

crzydg, 2020-07-28:
Hi @DevNulPavel @davidwernhart 

the trick is to read the complete instructions. Foolish to try to reset a T2 SMC when the device is sodding old.
Yes, I can confirm, resetting does clearly do the trick (and I can now safely use AlDente on a daily basis, and switch it off when traveling to customers.)

Thanks for your patience and help
stay healthy
-crzydg

closed 2020-08-06 by davidwernhart