AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #32: macOS Big Sur support?

elboboz3c, 2020-08-12:


muvo4k, 2020-09-27:
are you still into the project i would appreciate it !?

artooro, 2020-10-28:
Big Sur has optimized battery charging built in as a new feature diminishing the usefulness of this tool.

elboboz3c, 2020-10-29:
> Big Sur has optimized battery charging built in as a new feature diminishing the usefulness of this tool.

Actually it depends on your use scenarios. If you just want optimised battery charging then use the built-in one. Catalina also has that. But in my case, I use my MacBook plugged in most of the time, and after doing this for several month, my battery got destroyed. It got swollen and the actual capacity drastically reduced.
If you just rely on the built-in feature, which only reduces the charging level to 80-95% i think?, is not enough. The battery will get destroyed eventually. But if you use this app to adjust max charge to 50%, the battery will be perfectly healthy and if fact, my swollen battery returned to normal after doing this for several weeks, and the capacity even went back up a little bit. I thought I need to change the battery but now it seems unnecessary.

n808, 2020-11-13:
@elboboz3c Your description is very accurate. Apple's so called battery health management does not help in any noticeable manner if you keep your laptop plugged in most of the time. My MacBook Pro's battery is also almost destroyed from being at 100%. When I tried the battery health management introduced in 10.15, which is continued in Big Sur, it did absolutely nothing. With AlDente, I can keep the battery at 60% while plugged in, and the battery's steep degradation curve has stopped, and even slightly improved from "Service required" to "Good" status. I hope AlDente works in Big Sur, otherwise I will not upgrade. That's how important AlDente's functionality is for my usage scenario.

elboboz3c, 2020-11-13:
> @elboboz3c Your description is very accurate. Apple's so called battery health management does not help in any noticeable manner if you keep your laptop plugged in most of the time. My MacBook Pro's [...]

It seems to be working on big sur for me. I upgraded today and the max charge still holds at the limit I set using AlDente

will09122000, 2020-11-24:
Seems to be working for me too with new MacBook Air with the M1 chip.

davidwernhart, 2020-11-25:
@will09122000 Does it? I strongly suspected AlDente to not work on the new M1 Macs. Does the battery really stop charging? Best regards, David

will09122000, 2020-11-25:
Sorry my bad, I forgot to edit the post above when I realised it didn’t actually work. I could have a look and make a pull request if I get anything working?

soumahmed, 2020-12-08:
Can anyone please update it for Apple silicon?

Drallas, 2020-12-17:
On my Intel MacBook Pro with Big Sur it's working fine! Only it overshoots 80% and stays at 81%.

closed 2021-02-14 by davidwernhart