AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #33: After I uninstalled AIDente，macOS didn't stop charge battery even it‘s fully charged

msn1983aa, 2020-08-14:
After I uninstalled AIDente，macOS didn't stop charge battery even it‘s fully charged.

Did anyone can tell me how to uninstall AIDente in the correct way？

celsoazevedo, 2020-08-19:
Have you tried reseting the SMC?

https://support.apple.com/en-us/HT201295

gvillo, 2020-08-25:
same it's happening to me. I did SMC and nothing... how can we uninstall the helper?

gvillo, 2020-08-25:
Look for `com.davidwernhart.Helper` and erase both files (located under `/Library/PrivilegedHelperTools` and `/Library/LaunchDaemons`) and that's it.

Bables55, 2020-11-13:
I deleted those files and did an SMC reset but it doesn't improve.
Is there any other way?

andregriffin, 2020-12-11:
I ran into this same issue and came across this thread. I solved it by reinstalling Al Dente, making sure the max battery limit was set to 100% in the app, then going through the uninstall/SMC reset. It seems that uninstalling and resetting the SMC does not completely happen if a charging limit under 100% was in effect before uninstalling the app.

closed 2021-03-26 by MatthiasKerbl