AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #34: request: brew cask support

tshu-w, 2020-08-14:
As stated in the title, thank you.

peterlewis, 2020-09-05:
Have been working on a cask, but need to have a think about how to best integrate a suitable uninstall (i.e. should it be mentioned in a caveat that an SMC reset is required in order to fully uninstall?).

For now, if you're keen to get immediate cask support (albeit with an incomplete uninstall process), feel free to `brew tap peterlewis/homebrew` and then `brew cask install aldente`.

peterlewis, 2020-09-06:
> Have been working on a cask, but need to have a think about how to best integrate a suitable uninstall (i.e. should it be mentioned in a caveat that an SMC reset is required in order to fully uninstall?).
> 
> 
> 
> For now, if you're keen to get immediate cask support (albeit with an incomplete uninstall process), feel free to [`brew tap homebrew/peterlewis`](https://github.com/peterlewis/homebrew-peterlewis/) and then [`brew cask install aldente`](https://github.com/peterlewis/homebrew-peterlewis/blob/master/Casks/aldente.rb).

Apologies. I got my `brew tap` the wrong way round! I've edited my reply to include the correct command!

peterlewis, 2020-09-06:
Apologies! I got my `brew tap` wrong, twice! (I had already edited it, and got it wrong again!). I've updated my reply with the correct `brew tap` command!

tshu-w, 2020-09-07:
@peterlewis just a minor problem, and it's working as expected now. 
Maybe you should reset to 100% before uninstall?

closed 2020-09-07 by tshu-w

peterlewis, 2020-09-07:
> @peterlewis just a minor problem, and it's working as expected now.
> Maybe you should reset to 100% before uninstall?

Good to hear! Regarding reset to 100% - unless AlDente were to feature command-line support then I don't believe that it won't be possible to script that without external dependencies. For now I've added a caveat to the cask, instead.