AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #38: Almost working on MacBook Pro (13-inch, 2019, Two Thunderbold 3 ports)  (A2159)

jos-van-wezel, 2020-09-26:
Thanks David. 
Here's what happens on the mentions MacBook. 
With MacBook charger connected: After installation of AlDente and setting the new charging value to e.g. 80%, the battery stays at 100%. So it seems AlDente is not working since in the Readme it says "wait till the OS registeres the change" but the charging value stays at 100%. I was expecting the charge to drop until it reaches the setting of 80%. 
When I disconnect the charger and wait till the battery charging value drops to 85% I reconnect the charger and the charge value stays at 85%! Very good (see pic)
[image]. The charge stays at 85% even after I stop AlDente. That's fine since I'm mainly using my MacBook as desktop with the charger connected continuously. Of course I need to check if I can convince the SMC to charge to the full 100% again when I need the full capacity again. Hope this helps.

TheMazus, 2021-02-19:
Hi jos-van-wezel,
we have updated AlDente with new functionalities. The new version (2.1-alpha) has the beta feature "allow discharge", which does exactly what you asked for. Hope you like it.

jos-van-wezel, 2021-02-19:
Confirmed. On MacOS 11.2 v2.1 of AlDente works as described and as expected :-) Thanks a bunch.

jos-van-wezel, 2021-02-24:
Hi Matthias, David,

thanks for the heads up on the update. Very nice! 
I tried it immediately and indeed the battery now discharges to the set level and then stays there.

I found these 2 flaws
- when selecting options in the settings window the max charge slider above immed. moves to 100%. Its cosmetic but still annoying because if you try out another setting its not directly obvious the slider moved to max again.
- more serious is the fact that the machine goes to sleep with the lid closed and while an external monitor connected. Let me explain. My use case for alDente is the macbook connected to an external monitor, using the macbook with the lid closed (I think apple calls this clam shell mode). If I close the lid, the external screen briefly switches off and on again. This is all normal and I now I can use my Mac with the lid closed using an external keyboard.. The battery indicator tells me the battery is charging. The moment the charging stops (because the charge level reaches the set charge point) the machine goes to sleep. Not nice. I have to open the lid again because the keyboard does not wake up the machine. I can even force the sleep when I move the slider in AlDente below the current charge level and then close the lid. The machine immediately goes to sleep and cannot be waked unless I open the lid.

I’m not sure if I correctly call the above situation ’sleep’. Normally, in sleep state, I can wake up the machine with my external keyboard. In the sleep described for the situation above, I can no longer wake the machine. So I assume it has entered another state.

Configuration:
AlDente version 2.1alpha in BCLM key mode and in the ‘new’ mode (Charge Inhibit: true, Prevent Sleep: false, Inflow Enabled: true)
On MacOS 11.2.1 on a 2019 Intel MacBook Pro 13” 2 ports 
AlDente options: launch at login and allow discharge are checked. 

Hope this helps. Thanks for the effort.

beste groeten

Jos.
-- 
Steinbuch Centre for Computing (SCC)
KIT - Campus North
Hermann von Helmholtzplatz 1
76344 Eggenstein - Leopoldshafen
☏ +49 721 60826305
Building 449, Room 325
Orcid ID: 0000-0003-0175-6216

jos-van-wezel, 2021-02-25:
two more observations (on same config as written yesterday):
- after the machine has been put to sleep manually (shift-control-eject) it starts to charge the battery to maximum capacity. This can be seen after waking the machine.
- after wakeup the macbook does not start to discharge i.e. no longer discharges the battery (although checked in the settings of alDente.

best

Jos.
-- 
Steinbuch Centre for Computing (SCC)
KIT - Campus North
Hermann von Helmholtzplatz 1
76344 Eggenstein - Leopoldshafen
☏ +49 721 60826305
Building 449, Room 325
Orcid ID: 0000-0003-0175-6216

TheMazus, 2021-03-03:
Hi @jos-van-wezel,
I am glad you like AlDente!

Regarding the first flaw: This is a safety feature to avoid using different pathways to prohibit charging at the same time. However, we will definitely try to improve on this to make it more convenient.

Regarding the second flaw: Unfortunately, due to technical limitations, clamshell mode is not supported while "Allow Discharge" is enabled. At this point, we are not sure if we can do anything about it. However, when we find a way, we will definitely implement it.

Best regards,
Matthias

closed 2021-03-03 by TheMazus