AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #42: Max Battery Charge Defaults to 61

jumban, 2020-11-02:
Hi, regardless of which battery level I have set, the battery level resets to 61 after restarting the computer. Other than that, app works great.

2016 MacBook Pro 13" w Touch Bar
MacOS Catalina 10.15.7

Cheule, 2021-01-15:
Interestingly, I have a different issue.  On Big Sur v11.1, after reboot the battery level setting is always maxed out (103% when "Use MacOS scale" is set).

TheMazus, 2021-02-26:
Hi @jumban and @Cheule,
do you still have these issues with our new version AlDente 2.1-alpha?

Cheule, 2021-03-04:
> Hi @jumban and @Cheule,
> do you still have these issues with our new version AlDente 2.1-alpha?

Hi there! I recently switched from my Intel i9 2019 MacBook Pro to a brand new M1 MacBook Pro. 

So I’m happy to report that AlDente 2.1-alpha is working great, but unfortunately that doesn’t help you know if the issue was resolved with my 2019 i9 Intel machine.

TheMazus, 2021-03-04:
> Hi there! I recently switched from my Intel i9 2019 MacBook Pro to a brand new M1 MacBook Pro.
>
> So I’m happy to report that AlDente 2.1-alpha is working great, but unfortunately that doesn’t help [...]

I am glad AlDente is working fine on your new MacBook. Since no one else reported this issue on the new version, I think it's resolved.

closed 2021-03-04 by TheMazus