AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #44: MacOS Big Sur Compatibility

abshake96, 2020-11-13:
Hi,

First of all, thanks for this great tool! Just wanted to know if this tool will continue working in the Big Sur update?

If not, do you have plans to make an update? More than happy to help you test as per your convenience.

Additionally, if you're not planning to make the update, is there some kind of guidance you can offer as to the step we'd require to take. 

Thanks again!

diogoterremoto, 2020-11-13:
I've just upgraded from Catalina to Big Sur and AlDente seems to be working. 

[image]

[image]

davidwernhart, 2020-11-13:
I also just updated to big sur today, like @diogoterremoto it seems to work without any problems for me right now. Sadly, I cannot guarantee anything, but the outlook is good. I suppose that the underlying functionality of AlDente (SMC Key manipulating) will not stop to work even with future OSX versions. That being said, this method most likely will not work anymore on the new ARM Macs.

JGeek00, 2020-11-13:
I have a MacBook Pro 15 2019 with Big Sur and I confirm that it's working fine. Thanks to the developer for this app, this option should be native on macOS. I had my MacBook Pro 3 months connected to the charger because it was connected to two external monitors, and in that situation the battery drains so fast. After that 3 month, the battery is not working as it should be, it is jumping numbers on the percentage, and with the same use it has between one hour and one hour and a half less autonomy. So, again, thanks to the developer for this application.

togiberlin, 2020-11-19:
I can confirm. 
Works on my MB Pro 13" Early 2015 /w OS X Big Sur flawlessly.

hfsken, 2020-11-19:
Works fine on MacBook Pro (Retina, 13-inch, Late 2013) with macOS BigSur 11.0.1 (20B29)

brandonloong, 2020-11-23:
Working on my 2019 MacBook Pro 16" with Big Sur 11.0.1! Love this app, not sure why Apple can't just include this default like EVs do.

n-thumann, 2020-11-27:
Works on my MacBook Pro (15", 2018) running macOS 11.0.1 :v:

abshake96, 2020-12-05:
I've upgraded as well. Seems to be working great for me! Closing this issue now!

closed 2020-12-05 by abshake96