AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #47: blacked out icon in status bar 2020 base model Mac Pro 2 thunderbolt

hashsi99, 2020-11-14:
Hello, 

Trying to install this, but I'm getting a black logo in status bar.  Does this mean that this is not working?

If not, what's the proper way to uninstall?  First Mac....

thanks 
[image]

dtoniolo, 2020-12-02:
Hi @hashsi99, no need to worry: that's the default icon, there's nothing wrong. The original logo was colourful, but some users began complaining and the developer chose to go with black.

It should be working fine. Try to set a percentage (e.g. 80%) and see if changing stops at that level. Please note that in order for it to work you need to start _below_ the desired state of charge.

TheMazus, 2021-03-03:
> Hello,
> 
> Trying to install this, but I'm getting a black logo in status bar. Does this mean that this is not working?
> 
> If not, what's the proper way to uninstall? First Mac....
> 
> thanks
> [image]

Hi @hashsi99,
do you still experience this problem with the new version AlDente 2.1-alpha?

closed 2021-03-26 by MatthiasKerbl