AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #49: Interaction with new Battery Management Health functionality in 10.15.5

spockz, 2020-11-16:
Apple released new battery management in 10.15.5 that claims to extend the battery life: https://support.apple.com/en-us/HT211094.

This also discharges the battery every so often. Is there something known about the influence of setting the BMCL with Aldente on this? Should we not do this anymore and rely on Apple's improved battery management?

Oll0w4in, 2020-11-16:
In my oppinon definitly not. My personal usage is not something that can be learned from an algorithm. It is different every Day. So i deactivated it and load my Mac with Aldente to 60% and have enough battery if i need to use it a few hours without charging. 
Thus I don't know how good the Apple management is, but it is not helpful in my case. But nice that they think more about battery life, maybe a program like Aldente can be found in MacOS nativly sometimes :D

dtoniolo, 2020-12-02:
Most users are reporting that everything works fine even with both AlDente and Apple's battery management options enabled.

I prefer to keep only the former active, because Apple's feature is quite primitive: most of my use patterns can't be predicted in advance, so it is much simpler to just set the SOC value I want with AlDente and forget about it.

spockz, 2020-12-02:
I have been running with both Aldente and the macOS feature for two weeks now and can confirm that it will keep charge between 30% and my configured max of 75%.

The only annoying thing is that there is no warning that macOS is discharging and it already happened to me that I needed to go when my battery was down to 30%.

To improve this, could Aldente get some feature with a scheduled and announced drain to 30% which then resets to what was configured as the max charge?

closed 2021-07-05 by MatthiasKerbl