AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #52: Somone managed to run it on a M1 MBP with Big Sur?

Peebuddy43, 2020-11-20:
Somone managed to run it on a M1 MBP with Big Sur?

JiaaackY, 2020-11-21:
I tried on my m1 air. It doesn't work. I installed the helper tool and reboot my air but it still keeps charging.

davidwernhart, 2020-11-21:
@JiaaackY thank you for this information! I already suspected that it would not work on M1 macs since they have no SMC like the intel ones have. Maybe I will be able to find a similar solution for M1 macs in the future, but for now I don't have a device to test. Best regards, David

freestylemaster, 2020-11-25:
> @JiaaackY thank you for this information! I already suspected that it would not work on M1 macs since they have no SMC like the intel ones have. Maybe I will be able to find a similar solution for M1 macs in the future, but for now I don't have a device to test. Best regards, David

You can't imagine how many folks out there awaiting eagerly for the M1 version as this is the most amazing app you could ever install on your mac if you are careful about battery of your device. Love your work.

asd123cqp, 2020-11-26:
I believe SMC should still exist on ARM Macs, it's just that the key `BCLM` has been renamed or removed.

I complied the smc tool from [smcFanControl](https://github.com/hholtmann/smcFanControl/tree/master/smc-command) on an ARM Mac.
- `smc -l` still shows a ton of stuff
- `smc -l | grep BCLM` shows nothing (on an Intel mac it should give something like `BCLM  [ui8 ]  60 (bytes 3c)`)

DevNulPavel, 2020-11-26:
@asd123cqp 
Do you can show your output of `smc -l` command?

asd123cqp, 2020-11-26:
Sure, here's the `smc -l` output on a M1 Macbook Air: [smc-l-arm-mba.txt](https://github.com/davidwernhart/AlDente/files/5604908/smc-l-arm-mba.txt).

davidwernhart, 2020-11-26:
Thank you for dumping the smc keys @asd123cqp ! That is very interesting to me, I suspect Apple simply renamed some keys for the new M1 macbooks. From looking at your dump, I would guess that the key "BCMV" (maybe "battery charge max value"?) could be the right one. If you are brave enough, you could maybe try to write to it and see what happens. Although I must warn you that you can also brick your MacBook by writing to the wrong keys and I do not take responsibility if that would be the case.
Anyways, thank you for sharing your findings!
Best regards, David

Superjack78, 2020-11-26:
> Thank you for dumping the smc keys @asd123cqp ! That is very interesting to me, I suspect Apple simply renamed some keys for the new M1 macbooks. From looking at your dump, I would guess that the [...]

How could you brick it? Would resetting the SMC not fix an issue caused by this?

davidwernhart, 2020-11-26:
@Superjack78 because we don't know what those keys do for now. Maybe BCMV actually stands for "battery charge max voltage" and modifying it actually messes with the charging circuit, having the potential to do some permanent damage to the battery even if you reset the SMC afterwards.
The point is that we just don't know yet and there is always some risk attached to messing with the SMC. I got lucky with finding the original BCLM key, but I cannot guarantee that this is the case for everyone.

asd123cqp, 2020-11-26:
> From looking at your dump, I would guess that the key "BCMV" (maybe "battery charge max value"?) could be the right one.

@davidwernhart I doubt `BCMV` is the renamed `BCLM`. Its current value is `BCMV  [ui16]  23312 (bytes 5b 10)` and I am not sure how to make sense of this number. Maybe this is the new charge limit and it is now an absolute value instead of a percentage. Maybe it's totally unrelated. I don't know. 

There are two values that stand out to me though. I searched the dump for *keys that start with an "B" and have the value "100"* and found these two: `BMSC  [ui16]  100 (bytes 64)` and `BTRS  [ui8 ]  100 (bytes 64)`. Again, not sure what they stand for and might be totally unrelated. I don't know.

smoothdvd, 2020-11-27:
BTRS is for Bluetooth?

2018-08-07 09:16:29.919311+0100 0x7b       Default     0x0                  0      0    kernel: (IOBluetoothFamily) **** [IOBluetoothACPIMethods][SetBTRS] -- mACPIDevice->evaluateObject ('BTRS') successful -- took 23211 microseconds ****

[https://github.com/khronokernel/DarwinDumped/blob/b6d91cf4a5bdf1d4860add87cf6464839b92d5bb/MacBook/MacBook9%2C1/BootLog_Kernel/Kernel_Messsages_BootLog.txt](url)

thloh85-intel, 2020-11-30:
@asd123cqp  Wonder if you have managed to try out writing to BMSC?

timension, 2020-12-01:
I find it very strange that so many people (including me) want this feature (which is very sound, considering the sustainability etc.) but Apple cannot provide it natively. :-/

lasharor, 2020-12-01:
> I find it very strange that so many people (including me) want this feature (which is very sound, considering the sustainability etc.) but Apple cannot provide it natively. :-/

I believe it to be even more relevant now as with the increased battery life it becomes somewhat difficult to get this laptop to "healthy" battery levels in regular office use.

aomizu, 2020-12-01:
> I find it very strange that so many people (including me) want this feature (which is very sound, considering the sustainability etc.) but Apple cannot provide it natively. :-/

I believe apple doesn't want to spread the fact that keeping your battery at 60% is keeping it healthy. if too many people know it, the "user experience" might suffer, because then people are afraid of charging it up to 100%. they rather hide it as a "battery optimization" feature, that is out of control of the user

thloh85, 2020-12-04:
Did anyone tried out changing BMSC? I want to buy the new M1 MacBook, been using AlDente for a while, and wish to know if I can still use it for M1 MacBook.

stefandesu, 2020-12-08:
I'm slightly worried that this feature was removed from the M1 MacBooks. The support article "About battery health management in Mac notebooks" has two versions:
- Intel: https://support.apple.com/en-gb/HT211094
- Apple Silicon: https://support.apple.com/en-us/HT211832

The latter doesn't talk about the "battery health management feature" at all. 🤔

lasharor, 2020-12-08:
> I'm slightly worried that this feature was removed from the M1 MacBooks. The support article "About battery health management in Mac notebooks" has two versions:
>
> * Intel: [...]

[image]

stefandesu, 2020-12-09:
> [image]
> 
> M1 Macbook Air

What I meant is the option "Manage battery longevity" under Battery Health (see screenshot in the Intel article). That feature is missing on M1 Macs (at least on my MBA).

thloh85-intel, 2020-12-09:
Well as long as the hook to enable "Optimised battery charging" is there, there should be a way to limit the charge.
However, if the hook is now placed in the firmware (ie. ARM's SMC callback handles Y/N to optmize battery charging instead of the percentage), then we're out of luck.
I hope they didn't kill this.

@stefandesu Are you planning to try out changing BMSC's value?

stefandesu, 2020-12-09:
> @stefandesu Are you planning to try out changing BMSC's value?

I'm not willing to take the risk, sorry. 🙈

mattiasottosson, 2020-12-09:
I don't really get how the optimised battery charging works. Had that checked since day one on both my iPhone and M1 mac, but haven't seen it stay at 80% once. 

I use my macbook at work plugged in to a screen, so it stays at a 100% at all times now unfortunately. 

Maybe start a BMSC insurance campaign at [gofundme](https://www.gofundme.com/)? 🤓

celsoazevedo, 2020-12-10:
> I tried on my m1 air. It doesn't work. I installed the helper tool and reboot my air but it still keeps charging.

I had to set the % max and reboot my Intel MBP 16" _twice_ for the limit to be applied.

Not saying that AlDente works on the M1, but I would try set the limit and reboot more than one time. Also, if we set it to stop at 80%, sometimes it goes over that (usually 1 or 2% above the value we set).

Evertt, 2020-12-10:
No it just doesn't work on M1 MacBooks.

>

thloh85-intel, 2020-12-11:
> No it just doesn't work on M1 MacBooks.
> […](#)

It won't work without proper changes definitely. As per @asd123cqp's check, the BMCV SMC value is no longer there, thus we need to look for the proper value to change (if it exists)

Evertt, 2020-12-11:
Yes I'm aware, I read all of that. I hope the value still exists under a
different name, because this tool is super useful.

davidwernhart, 2020-12-11:
> I don't really get how the optimised battery charging works. Had that checked since day one on both my iPhone and M1 mac, but haven't seen it stay at 80% once.
>
> I use my macbook at work plugged in [...]

This sounds like a good idea, but I honestly doubt to get enough backers to afford a ~1100€ M1 MacBook Air to experiment with. I guess it's much more likely that someone who already owns one finds the right key in the meantime.

Best Regards,
David

thloh85, 2020-12-12:
Honestly, I suspect they moved it to firmware level callback (ie. the ARM SMC callback) instead of the SMC value. 
I've been working on ARM bootloader and are familiar with SMC callback's functionality, and this is one of the implementation that I personally would put in as an SMC call. As for how it is called, what value, it really is up to the kernel level driver. In Linux, it is open source, thus is pretty open and accesible how one can set it, but in MacOS, not so much. However, Darwin kernel is open source, so, in that department, we might find something :)

Ryanfsdf, 2020-12-13:
I've figured out some more clues in case anyone is willing to risk their their laptop.

When I first got my M1 MBA, there were 4 values in the SMC which had a value of 100. These were BMSC, BNSC, BTRS, and BUIC.

Upon checking now, I see that BMSC and BNSC are both at 99 instead of 100.  They seem to be related in some way, and the fact that they dropped to 99 seems to suggest that they are percentages.

However, even though both of these values are 99, my laptop still charges all the way up to 100%. This may not mean much though, because I've noticed on all my MacBooks that even if my laptop battery is ~97% full based on actual current battery capacity divided by the maximum capacity, the laptop doesn't charge beyond that and the battery percentage reported is still 100%.

thloh85, 2020-12-13:
This could also mean that the BMSC is the max battery percentage (ie. your
battery health).

Ryanfsdf, 2020-12-13:
> This could also mean that the BMSC is the max battery percentage (ie. your battery health).
> […](#)

I don't think it is, because my max battery capacity is at 4420mAh and the design capacity is 4382mAh.

DevNulPavel, 2020-12-13:
Maybe BMSC stands for "Battery Max Supported Charge"?

thloh85, 2020-12-14:
> > This could also mean that the BMSC is the max battery percentage (ie. your battery health).
> 
> I don't think it is, because my max battery capacity is at 4420mAh and the design capacity is 4382mAh.

Yeah, but from @asd123cqp's log, the value is pretty different (BMSC == 100, BNSC == 92). I couldn't make out what's the actual value :(

smoothdvd, 2020-12-14:
@thloh85 my smc data:
capture 1 (stop charging at 70%):
```
BMSC  [ui16]  70 (bytes 46) 
BNSC  [ui16]  37 (bytes 25) 
```
capture 2(stop charging at 66):
```
BMSC  [ui16]  70 (bytes 46), 
BNSC  [ui16]  37 (bytes 25)
```
capture 3(stop charging at 74%)
```
BMSC  [ui16]  74 (bytes 4a)
BNSC  [ui16]  37 (bytes 25)
```

[image]

suppose it is the BMSC means MAX value in some period when charging is stopped (Battery Max Stopped Charge) and BNSC means MIN value in some period when charging is started(Battery Max Started Charge).

thloh85, 2020-12-14:
> @thloh85 my smc data:
> capture 1 (stop charging at 70%):
> 
> ```
> BMSC  [ui16]  70 (bytes 46) 
> BNSC  [ui16]  37 (bytes 25) 
> ```
> 
> capture 2(stop charging at 66):
> 
> ```
> BMSC  [ui16]  70 (bytes 46), 
> BNSC  [ui16]  37 (bytes 25)
> ```
> 
> capture 3(stop charging at 74%)
> 
> ```
> BMSC  [ui16]  74 (bytes 4a)
> BNSC  [ui16]  37 (bytes 25)
> ```
> 
> [image]
> 
> suppose it is the BMSC means MAX value in some period when charging is stopped (Battery Max Stopped Charge) and BNSC means MIN value in some period when charging is started(Battery Min Started Charge).

Yeah, this makes a lot of sense! Thanks for trying out, this makes me instantly buy the Macbook Air!
I guess what we'll need to do is to integrate the changes back into AlDente. Wow thanks a lot @smoothdvd for trying this out and reporting back, I guess this is where open source benefits most!

thloh85, 2020-12-14:
> > @thloh85 my smc data:
> > capture 1 (stop charging at 70%):
> > ```
> > BMSC  [ui16]  70 (bytes 46) 
> > BNSC  [ui16]  37 (bytes 25) 
> > ```
> > 
> > 
> > capture 2(stop charging at 66):
> > ```
> > BMSC  [ui16]  70 (bytes 46), 
> > BNSC  [ui16]  37 (bytes 25)
> > ```
> > 
> > 
> > capture 3(stop charging at 74%)
> > ```
> > BMSC  [ui16]  74 (bytes 4a)
> > BNSC  [ui16]  37 (bytes 25)
> > ```
> > 
> > 
> > [image]
> > suppose it is the BMSC means MAX value in some period when charging is stopped (Battery Max Stopped Charge) and BNSC means MIN value in some period when charging is started(Battery Min Started Charge).
> 
> Yeah, this makes a lot of sense! Thanks for trying out, this makes me instantly buys the Macbook Air!
> I guess what we'll need to do is to integrate the changes back into AlDente.
> 
> Thanks a lot @smoothdvd for trying this out and reporting back, I guess this is where open source benefits most!

I'll read up AlDente's code to see if I know how to change it. I'm not very familiar with app development as I'm a low level bootloader engineer. 

But I guess I can navigate it around if @davidwernhart doesn't have the time to do so.

Evertt, 2020-12-14:
Hmm, I don't see yet how this helps. The way I understand it, the values
you found function like some kind of log of what the last max value was at
which you stopped charging? Isn't it? But we don't need a log-value, we
need a value that dictates when the system will choose to stop charging.
Don't you think?

Or did I misunderstand something? Did you in fact first change those values
to some specific values and then see what would happen when you let your
macbook charge?

thloh85, 2020-12-14:
Ahh I must have misunderstood then. I guess you interpret this correctly and I misinterpreted it.

Ryanfsdf, 2020-12-14:
I guess we can rule out BMSC and BNSC then. That leaves BTRS and BUIC. I think it was mentioned before that BTRS is a new value. Does anyone know if BUIC is present on an Intel MacBook? Also, is there anyone with a BTRS or BUIC value that isn't 100?

smoothdvd, 2020-12-14:
@Ryanfsdf BUIC means current battery percent.

freestylemaster, 2020-12-14:
> @Ryanfsdf BUIC means current battery percent. And on my M1 MacBook Air, only BTRS value is 100 now.

Would you mind sharing how I can extract SMC data?

smoothdvd, 2020-12-14:
@freestylemaster https://github.com/davidwernhart/AlDente/issues/52#issuecomment-734028300

Frederik441, 2020-12-16:
Can read value 100 on M1 Mac with key BTRS. I would like to try change it but can't. if I do "smc -k BTRS -w 65" the command is accepted but when reading it again it show still 100. Also tried to change it from recovery - terminal, but got wrong architecture failure from the smc command. Any clues?

smoothdvd, 2020-12-25:
Finally, I saw this battery optimization status.
[image]

freestylemaster, 2020-12-25:
> Finally, I saw this battery optimization status.
> 
> [image]
> 
> 
> 
> BUIC value is 80.

Wonderful, thanks. You even get the option to charge it to full. That's really interesting.

I hope someone can figure out how to implement it to the app.

Frederik441, 2020-12-25:
wow, please tell me how do you set the value 80 in BUIC?

Frederik441, 2020-12-25:
No, BUIC shows the actual battery charge procent. But how did you set the battery charging on hold?

davidwernhart, 2020-12-25:
Oh, okay. I also got a little too excited about this for a moment.
Anyway, as soon as someone manages to find the right key, I will make sure to have an update ready to go.
Happy Holidays,
David

Evertt, 2020-12-25:
@smoothdvd just to check, did you just get this because you checked the checkbox in system preferences? Or did you get this because you experimented with setting SMC values?

smoothdvd, 2020-12-26:
@Evertt Yes, I checked the checkbox in system preferences.

freestylemaster, 2020-12-26:
BUIC is the current battery percentage as commented before.

[image]

freestylemaster, 2020-12-27:
Captured smc while battery was on hold by Apple's own Battery Health Management system.

This might be what we are looking for. Anyone knows if we had these 2 keys on Intel's?

[image]

Frederik441, 2020-12-27:
-> freestylemaster, would you like to send us a smc dump after the charge on hold stops? So vi can do a diff.

freestylemaster, 2020-12-27:
> -> freestylemaster, would you like to send us a smc dump after the charge on hold stops? So vi can do a diff.

Here's the smc log with no hold - at 100%

[image]

smoothdvd, 2020-12-28:
Maybe all SMC in Apple Silicon Mac is "virtual" and "readonly".

thloh85-intel, 2020-12-30:
I guess we'll just have to wait for Apple to release the XNU kernel source, then we'll be able to take a look if they do expose it via ARM SMC callback

thloh85, 2021-01-03:
Speaking of which, did anyone tried writing to SBAR?

Frederik441, 2021-01-03:
Tried now, it looks like a down counting counter when on battery and up counting on power. Can't change the value with write another value to it.

Harry1993, 2021-01-05:
Could it be `BLTA`? I ran `./smc -l | grep "20 ("` and it gave me
```
BLTA  [si8 ]  -20 (bytes ec)
...and some other results...
```
However, when I turn off `Optimized battery charging`, `BLTA` is still `-20` (maybe because there is another boolean SMC value for whether or not the option is turned on?).

F3000, 2021-01-08:
When I understand the thread until now right , it doesent run on M1 chips ?

[217 more replies]

closed 2021-02-14 by davidwernhart