AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #53: Options to check/uncheck inside Battery Settings on Big Sur?

joshuarobs, 2020-11-20:
Do we uncheck both `Optimised battery charging` and `Battery Health... > Manage battery longevity` while having Aldente run at the same time?

JasSingh685, 2020-11-20:
I leave mine on. I don't think it will cause any issue. I believe macOS will lower the charge level over time, as the battery ages, whereas AIDente you limit the charge level manually before macOS does anything.

joshuarobs, 2020-11-21:
@JasSingh685 Which one do you leave on? Both?

JasSingh685, 2020-11-21:
> @JasSingh685 Which one do you leave on? Both?

Yeah I keep it on default which is both set to on, seems to be ok and still charges to 100% when I need it to.

joshuarobs, 2020-11-21:
@JasSingh685 
I'm confused. It charges to 100%, but isn't the point of AlDente to set the limit to anything below that, e.g. 50%, 60%, 70%, 80%?

I wish there was more information on this app, its so confusing and vague and its hard to make the right decision

JasSingh685, 2020-11-21:
No sorry my mistake, I meant to say that I can set the app to charge the battery to 100% if I **need** to, but normally I set it to around 75%, so yes you can set to whatever you like!

joshuarobs, 2020-11-21:
Thanks for the clarification

closed 2020-12-11 by davidwernhart