AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #54: The battery just keeps charging

ccat903, 2020-11-20:
The battery just keeps charging, it doesn't stop at 80% as I set it.

[image]

JasSingh685, 2020-11-20:
Have you tried rebooting your Mac?

heartbleeded, 2020-11-22:
> Have you tried rebooting your Mac?

Yes. Doesn't work.

ccat903, 2020-11-22:
It does not work.

smoothdvd, 2020-11-24:
Which model do you use?

JasSingh685, 2020-11-24:
By any chance, have you tried draining the battery a little? Say, below 50%, and see if it then stops at the 80%? 

From what I understand, it doesn’t allow for discharge while plugged in

ccat903, 2020-11-26:
I have a 2019 13' mac book pro, I discharged it to like 70% and then plugged it back in.

ccat903, 2020-11-29:
Oh.. Looks like it works now... I just had to discharge to lower then 50% and then charge. weird

closed 2020-11-29 by ccat903