AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #55: Doesn't work on Big Sur with MBP 16"

richardsimko, 2020-12-05:
I'm trying to troubleshoot this app not working at all on my MBP 16" from 2019. Regardless of what I set the setting to it seems to do nothing. Considering other people are reporting it working on Big Sur I'm presuming there's something it's conflicting with on my system. Is there any logs or something like that I can check?

rootedetc, 2020-12-09:
Same for MBP 15" running Big Sur.

kodifiera99, 2020-12-09:
The dev literally states Big Sur is not compatible. Big Sur is a buggy mess, either downgrade, or you won't be able to use the app..

richardsimko, 2020-12-09:
> The dev literally states Big Sur is not compatible. Big Sur is a buggy mess, either downgrade, or you won't be able to use the app..
It's great that they state that, but last I checked this was an open source project, meaning it's possible for users to improve it and expand compatibility. However the first step would then be to understand what's going wrong and a good place to start would be logs - Hence my question about where they might be.

celsoazevedo, 2020-12-09:
It's working for me (same laptop running Big Sur), but while before I could just set the limit and it would be applied immediately, on Big Sur I have to set the limit and reboot. It didn't work at first, but on my second try, it did.

I have gatekeeper disabled (for unrelated reasons), I don't know if it has any influence over this though.

Limited to 85%:

[image]

richardsimko, 2020-12-09:
Interesting, I'll try another reboot to see if that helps!

rootedetc, 2020-12-09:
I jumped to conclusion, AlDente works after setting max charge and rebooting the machine.

richardsimko, 2020-12-10:
Yep, another reboot solved it for me as well!

closed 2020-12-10 by richardsimko