AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #57: Consider making this app available from homebrew-cask

R0LA1mRifcF8yAkk, 2020-12-09:


cwerner1, 2020-12-10:
I did an pull request @https://github.com/Homebrew/homebrew-cask/pull/95182 
as in the readme from aldente stated only for catalina.

cwerner1, 2020-12-10:
@davidwernhart would you please assing me this issue?
Thank you.

R0LA1mRifcF8yAkk, 2020-12-11:
Thank you, I hope this will get merged soon! Although, official Big Sur support would be very nice as well. I haven't had any issues so far at least.

cwerner1, 2020-12-11:
My Pull Request https://github.com/Homebrew/homebrew-cask/pull/95182 was declined because AlDente only supports Catalania, as stated in the readme.

I will reopen my pull request when AlDente gets updated to support Big Sur or this pull request from AlDente gets merges https://github.com/davidwernhart/AlDente/pull/48

davidwernhart, 2020-12-11:
It does work on Big Sur! At least for me and many other users.

@cwerner1 Since I updated the readme, maybe they will accept your pull request now.

Anyways, great work!

Best regards,
David

R0LA1mRifcF8yAkk, 2020-12-11:
@davidwernhart Can you detail the steps? Because I have 80% defined in the app and my computer charges to 100%. MBP 15" 2015

R0LA1mRifcF8yAkk, 2020-12-21:
The readme contains conflicting information at the moment.
> Currently ONLY works with macOS 10.15 Catalina.

In the FAQ:
> Allthough AlDente was written and compiled in Catalina, it works just fine on MacOS Big Sur.

simmac, 2021-01-14:
I added a cask formula for AlDente, already merged: https://github.com/Homebrew/homebrew-cask/pull/98144

Sorry @R0LA1mRifcF8yAkk, I only saw your PR after I already created mine, didn't intend to snatch away something from you

R0LA1mRifcF8yAkk, 2021-01-14:
@simmac  thank you! no worries, i don't care about credit, as long as stuff gets done it's all good

closed 2021-02-14 by davidwernhart