AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #58: Confirm working with MacBook Pro Retina 15 Mid 2015 Catalina 10.15.7

jcr-, 2020-12-11:
As some others have noted, did not work on initial launch after install. I set battery charge to 70 percent and setting appeared to be ignored.

However, a quit-and-relaunch showed the slider set to 103%, and adjusting back to 70 then worked as expected.

One caveat: Set to "Use MacOS Battery Scale" seems to result in setting being off by -1%. Obviously not significant, but setting to i.e 70% seems to cause iStat battery menubar to stop charging and stick at 69% (Nice.)
Coconut battery displays (presumedly non-MacOS battery scale) charge of 66% under same setting.

Anyway, works great and more than close enough for intended purpose, and this app would have saved me needing a new battery because my Mac Book is usually docked on a charger and connected to a larger monitor, especially these days.

joostk90, 2021-02-18:
Same here since 6 December 2020. I am using a MacBook Pro (Retina, 15-inch, Mid 2015) running Catalina 10.15.7 (19H2). AlDente is behaving as expected after a few installation attempts and rebooting.

I am controlling the charge by setting the level using AlDente to 50%, when it is reached, I set it back to 30% so the battery is 'left untouched' and the MacBook is powered by AC only, without charging. The battery's charge will decrease slowly to 30% in about two days depending on higher CPU load which - every now and then - requires a bit more power than AC alone, I think.  
I estimate this saves the battery from degrading.

TheMazus, 2021-02-18:
Hi guys, glad to hear AlDente is working for you.
Also joostk90, we are working on implementing this feature, so you don't have to do it manually every other day.
Greeting,
Matthias

closed 2021-02-24 by TheMazus