AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #59: Allow hiding of menu bar icon

SheaHolden, 2020-12-12:
Not an "issue" but it would be nice feature. Allow for the menu bar icon to not be displayed. Once the user sets their desired max charge I imagine it will be rarely changed. 

Maybe make a simple toggle in the settings hide the menu bar icon, and "launching" the app again (opening it normally by clicking on the app icon) makes it re-appear? or something along those lines. I have not created any mac apps before so i am not sure how easy it is to do.

TheMazus, 2020-12-13:
Hey Shea, a quick workaround for this problem is to simply exit AlDente after you have set the desired battery percentage. The maximum battery percentage will be saved in the system until you open AlDente and change the value again.

dr3mro, 2020-12-13:
You can simply check it app is running then show setting if not then launch it

joshuarobs, 2020-12-17:
I hide it with Bartender like all my other icons

bigunyak, 2021-02-21:
Adding my vote for this small feature as well.
The app indeed doesn't need to be accessed often once desired settings have been configured, and taking space in the menu bar seem unnecessary.

bruno-robert, 2021-10-29:
> Hey Shea, a quick workaround for this problem is to simply exit AlDente after you have set the desired battery percentage. The maximum battery percentage will be saved in the system until you open AlDente and change the value again.

quitting aldente on macOS Monterey 12.0.1 (21A559) doesn't seem to save the setting.
My battery will stop charging if it's open (in menubar) and continue charging the moment I quit aldente

MatthiasKerbl, 2021-10-30:
> > Hey Shea, a quick workaround for this problem is to simply exit AlDente after you have set the desired battery percentage. The maximum battery percentage will be saved in the system until you open AlDente and change the value again.
> 
> quitting aldente on macOS Monterey 12.0.1 (21A559) on a M1 Macbook Pro doesn't seem to save the setting. My battery will stop charging if it's open (in menubar) and continue charging the moment I quit aldente

Hi @bruno-robert,

Due to the new hardware of the M1 MacBooks, we use a different pathway to control the charging behavior. Unfortunately, it's not possible anymore to use this workaround.

Cheers,
Matthias

MatthiasKerbl, 2021-12-11:
Hi everyone,

We just released a new version of AlDente Pro which has a new option to disable the menu bar icon completely.

Cheers,
Matthias

closed 2021-12-11 by MatthiasKerbl

MikixIT, 2022-10-16:
"We just released a new version of AlDente Pro which has a new option to disable the menu bar icon completely."
I HAVE TO PAY FOR HIDING AN ICON?

AldrinSalazar, 2024-06-11:
> Hi everyone,
> 
> We just released a new version of AlDente Pro which has a new option to disable the menu bar icon completely.
> 
> Cheers, Matthias

Paying to hide an icon is so low ... I was considering buying it but i won't give my money to idiots.

StianOvrevage, 2024-08-15:
It IS possible to hide it in the free version, ref https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/501