AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #63: Big sur 11.1 doesn't work

MuReinhard, 2020-12-19:
MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports)

MuReinhard, 2020-12-19:
Version 1.0 is working

kamranzafar, 2021-01-07:
It actually works on Big Sur if you uncheck "Use MacOS battery scale" in the settings.

petter1, 2021-01-20:
I try it today with this Mac. It charged today to 100% with AlDente 1.2 and MacOS battery scale enabled. Now I let it discharge while I use my Mac and will then try alDente 1.2 with MacOS battery scale disabled.
[image]

TheMazus, 2021-03-14:
> I try it today with this Mac. It charged today to 100% with AlDente 1.2 and MacOS battery scale enabled. Now I let it discharge while I use my Mac and will then try alDente 1.2 with MacOS battery scale disabled.
> 
> Edit: AlDente did not stop my Mac from charging above 80%
> 
> [image]

Hi @petter1,
have you tried the latest version of AlDente? This one should work for you as well.

Matthias

petter1, 2021-03-16:
@MatthiasKerbl everything works fine. I only had to restart the app some times back then in order to get it working 😇

closed 2021-03-16 by MatthiasKerbl