AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #68: Magsafe charging light orange

adrian70000, 2020-12-29:
Hello,

Thanks for making this awesome tool. I use Macbook Air 2015 with a Magsafe charger. App works great and stops charging the battery but Magsafe light still remains orange instead of turning green. Is that normal?

aomizu, 2021-01-01:
yes it's normal. it just means that the battery is not fully charged. but there is still power running through the charger, which is used up by the components instead of charging the battery

Cyberdroid1, 2021-01-02:
I think making it green should be better, or, as an option.

Cheule, 2021-01-15:
> I think making it green should be better, or, as an option.

I could be mistaken, but my assumption is that the green/amber light in the old MagSafe connectors is triggered by battery voltage, not the OS.  So, if your battery SOC is less than 8.4v (for a 2 cell config) it's likely to show as amber on the MagSafe.  FWIW, there's no good way to know how much mAh is left in a battery, it's always a guesstimate based on voltage readings.

Cyberdroid1, 2021-01-15:
From https://github.com/godly-devotion/charge-limiter

[image]

I hope this clarifies :D

Cheule, 2021-01-15:
I stand corrected!  Thanks for posting.

Cyberdroid1, 2021-01-15:
@davidwernhart Can you please implement this? Sorry for pinging 😅

hpapagaj, 2021-01-16:
Following. This would be great. I also have orange light, but the app works great.

[image]

Fenderman2, 2021-12-22:
Any update on this? It would be great if this could be added (max optional) in one of the next releases. It seems like the codebase is already there?
https://github.com/godly-devotion/charge-limiter

Thanks in advance!

vvalzon, 2022-05-20:
This would make a huge difference! The ability to see if your charge is complete without opening the computer is incredibly convenient, and this issue is one of the only things that makes me waffle about using Al Dente!

cmalex23, 2022-08-05:
So any reason why the code from the `charge-limiter` app wasn't merged?

MatthiasKerbl, 2023-01-01:
Hi everyone,

We have just released AlDente 1.20 with a new beta feature called "Control MagSafe LED" which will finally properly indicate if the Charge Limit is reached. You can read more about this feature on our blog here: https://apphousekitchen.com/feature-explanation-control-magsafe-led/

Cheers,
Matthias

MatthiasKerbl, 2023-01-12:
Hi everyone,

Since we have finally implemented the features requested, I will close this issue.

Cheers,
Matthias

closed 2023-01-12 by MatthiasKerbl