AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #71: When can we expect an update for M1 MacBooks?

LeshawnG, 2021-01-05:
One of the most simple and important softwares for the Mac. Patiently waiting for it to be updated to work with Apple Silicon

suhuaixing, 2021-01-06:
So before there's an app, what should I do. Unplug it at roughly 80%, and plug it at 20%. Or always plug in. I'm at office with an adapter all the time.

weiliddat, 2021-01-07:
I would suggest keeping it plugged in and enable Battery Health Optimization. macOS should slowly decrease the max charge over time if it's constantly plugged in.
My old MBP13 was always plugged in (before AlDente and Battery Health were around) and still had 103% max charge and only 30+ cycles used after 2 years of use.

hgo11, 2021-01-19:
I hope there will be a working version available soon. I'm using my M1 mostly plugged in and try to run it on battery every other day. So far I'm at 14 cycles and already see a capacity drop by 4%. The built in battery management only kicked in once - delayed full change by one full hour - that's worse than a bad joke.

davidwernhart, 2021-01-19:
Sadly, there is not much I can do about it.
I do not have an Apple Silicon Mac nor can I afford one right now.
As soon as someone finds a solution that works, I will build it into AlDente immediately, but until then we have to wait until someone with the hardware finds the right key.

hgo11, 2021-01-19:
thanks a lot for all the efforts put into getting it working.

For all the folks waiting - here is the workaround I just tried and it seems to be working:
- discharge your laptop to cca 50%
- connect it to a phone charger cca 5v 1A - 2A 
- it will run on power from adapter but will stop battery charging

thu-eri, 2021-01-21:
> thanks a lot for all the efforts put into getting it working.
>
> For all the folks waiting - here is the workaround I just tried and it seems to be working:
>
> * discharge your laptop to cca 50%
> * [...]

Hey,
Do you mean to discharge until 50% and then plug it in with an additional phone charger?
BR

petter1, 2021-01-21:
Yes, the m1 laptop does not touch the battery, if you charge with a Powerbrickadabter that gives less then (xx)Watt. According to hgo11, 10W is enough to run the m1 laptop without using energy from the battery

hgo11, 2021-01-21:
So after experimenting the last days with this - during normal use (browsing, podcasts.. etc) I have a 5V 1.7A power brick plugged in - the battery is not charging and the computer is running from the power brick just fine. As soon as you start using more intensive applications (e.g. gaming) the 10W supply is not enough and additionally to using the 10W from the brick the M1 will also use some battery. 

Also - do not leave the 10W plugged in during night - when computer is in sleep it will activate charging even from the 5V 2A supply.

stanley-su, 2021-02-08:
> thanks a lot for all the efforts put into getting it working.
>
> For all the folks waiting - here is the workaround I just tried and it seems to be working:
>
> * discharge your laptop to cca 50%
> * [...]

Specifically which charger and cable are you using? I own a 5W Apple power adapter, but it's USB-A so it doesn't seem to charge the M1.

TheMazus, 2021-02-14:
The time has come: AlDente for Apple Silicon (M1 MacBooks) is out now. We thank everyone who has helped us here in the forum and we hope you like it. That's not all, we have also started working on a pro version of AlDente with advanced features several weeks ago. Since we have put a lot of hours into developing the app, it will cost a little. However, there will always be this free and open-source version of AlDente updated with the new M1 SMC keys.  If you have any feature requests or ideas for AlDente Pro, let us know. If you want to get notified, when AlDente Pro comes out, just enter your email at https://apphousekitchen.com/aldente/ In the meantime, stay safe and keep your battery fresh. David & Matthias

closed 2021-02-14 by davidwernhart