AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #72: Weird logging messages about com.davidwernhart.AlDente-LaunchAtLoginHelper

kandrei, 2021-01-06:
Hi,

I have the following messages in system.log:

Jan  6 09:40:44 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[73475]): Could not find and/or execute program specified by service: 107: Malformed bundle: com.davidwernhart.AlDente-LaunchAtLoginHelper
Jan  6 09:40:44 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[73475]): Service exited with abnormal code: 78
Jan  6 09:40:44 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper): Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
Jan  6 09:40:54 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[73483]): Could not find and/or execute program specified by service: 107: Malformed bundle: com.davidwernhart.AlDente-LaunchAtLoginHelper
Jan  6 09:40:54 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[73483]): Service exited with abnormal code: 78
Jan  6 09:40:54 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper): Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
Jan  6 09:41:04 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[73484]): Could not find and/or execute program specified by service: 107: Malformed bundle: com.davidwernhart.AlDente-LaunchAtLoginHelper
Jan  6 09:41:04 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper[73484]): Service exited with abnormal code: 78
Jan  6 09:41:04 ziMac com.apple.xpc.launchd[1] (com.davidwernhart.AlDente-LaunchAtLoginHelper): Service only ran for 0 seconds. Pushing respawn out by 10 seconds.

These are repeating for ever.

I did try to reinstall the helper or tick/untick the "Launch at login" checkbox but nothing helped. BTW, unchecking "Launch at login", exiting the app and starting it again shows the checkbox checked, even if previously it was unchecked.

Is there any solution to get rid of the unwanted error messages? It seems like the "LaunchAtLoginHelper" is somehow malformed. The OS version is Big Sur 11.1, AlDente version 1.2.

Thanks.

TheMazus, 2021-02-24:
Hi @kandrei,
have you tried our new version 2.1 alpha?
Maybe it will fix your error message issue.
Best regards,
Matthias

kandrei, 2021-02-24:
Hi @TheMazus,

Thanks for letting me know about the new version. I've installed it, removed the old version and after that everything went back to normal.

closed 2021-02-24 by kandrei