AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #73: Difference between AlDente an BigSur optimized battery

F3000, 2021-01-08:
Which is better when MacBook is 95% of the time at power cord ?

alexw9988, 2021-01-11:
I have mine plugged in all day and MacOS just kept charging to 100%, because it couldn't really "learn" any charging habit. 
AlDente works great, just make sure you leave "Use MacOS battery scale" option unchecked if you're running Big Sur.

F3000, 2021-01-11:
Ok . Do you use intel or M1 chip ?

alexw9988, 2021-01-11:
Intel

Drallas, 2021-01-31:
What does "Use MacOS battery scale" exactly do?

TheMazus, 2021-01-31:
Hey Drallas,

Normally, there is an offset of a couple of percentage points between the systems SoC and the shown SoC. The feature "Use MacOS battery scale" tries to decrease this offset. In a nutshell, when this feature is activated the SoC you set in AlDente will be closer to the "real" SoC shown in MacOS.

Best regards,

TheMazus

emadpres, 2021-03-07:
@alexw9988 Based on what @TheMazus said, it seems we should have `Use MacOS battery scale` checked as it's a good thing. Why did you recommend turning it off?

emadpres, 2021-03-07:
And by the way, where is the `Use MacOS battery scale` option? I can't find it in `2.1`.

TheMazus, 2021-03-07:
Hi @emadpres,
we removed the "Use MacOS battery scale" option in AlDente 2.1 since we use a different pathway to get as close to the MacOS battery scale as possible.

Best regards,
Matthias

closed 2021-03-14 by TheMazus