AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #75: Doesn't seems to work on MacBook Pro 13,3 with macOS 11.1

dezzeus, 2021-01-14:
I've tried it on my MacBook Pro 13.3 (15" 2016 w/ TouchBar) with MacOS 11.1 (because also the native 80% solution doesn't seems to work).

I've started with a 75% charge on the machine and set a 77% level in the app (having read that it could be shown with a 3–5% difference); I've then waited some minutes (about 5/10) and then I plugged in the charger… what happened is that the Mac ignored the app and went up to 83% (79.x% as reported by coconutBattery) and was still going on…

Side note: when opened for the very first time, the app's slider was set to 0 (zero); maybe you should provide a better default (100% ?).

GZaccaroni, 2021-01-15:
Have you seen #63 ?

dezzeus, 2021-01-15:
Yes.

Interestingly, another (CLI) software did the job: [this one](https://github.com/zackelia/bclm); maybe you can investigate the differences in order to improve this one.

TheMazus, 2021-02-14:
Hi dezzeus, we have just released a new version of AlDente. Maybe this one will work for you.

closed 2021-03-14 by TheMazus