AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #87: 103% charge issue where AlDente shows correct battery level (not 3% offset)

CompilerStyler, 2021-01-29:
When I set AlDente to 80% I get exactly 80% charge per the MacOS battery indicator. This worries me because AlDente slides to 103% due to the 3% offset issue - does this create a hazard on systems that no longer require the offset (e.g. the laptop will try to actually overcharge itself)

System: 
MacBook Air "Core i3" 1.1 13" (Scissor, 2020) 1.1 GHz Core i3 (I3-1000NG4)
Big Sur 11.1

TheMazus, 2021-01-29:
Hey CompilerStyler,
this issue should only appear after the first restart, afterward, it should stay at 100% as well. 
Even if you leave it at 103% it doesn't overcharge, so no worries.

CompilerStyler, 2021-01-29:
Ah I see that now - thank you for checking!

closed 2021-01-29 by CompilerStyler