AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #91: battery drain while sleeping

ryansyrl, 2021-01-31:
Hey guys, i installed al dente on my mba 2020, after i installed it i leave my macbook sleep overnight, the morning i wake up my battery lost 4%.. i did SMC reset, the result is still the same, my battery lost 1% of battery charge 1hr sleep. before i installed al dente if i leave it sleep overnight, not even 2% from the current charge.. have anyone have the same issue? what should i do? any suggest?

pedrocatalao, 2021-02-01:
It depends on which processes you have running before putting the mac to sleep, sometimes I get no battery drain, other times I get 15%, it's not related to Al Dente

ryansyrl, 2021-02-01:
I thought so, but the the weird thing is, before i installed it, while it sleeps overnight barely loss 1% from current charge. With the almost the same multi tasks running before it sleep..

TheMazus, 2021-02-14:
Hi ryansyrl, do you still experience a higher battery drain while sleeping? We have just released a new version of AlDente, maybe this one works better for you.

Lamedouard, 2021-06-09:
Well for me I lose 15% per night. I have installed the 24 Mar release version on my MBP late 2013 running bigSur. Does anyone has the same issue?

closed 2021-07-05 by MatthiasKerbl

fragment75, 2021-10-31:
i loose 40% in under an hour while sleeping

pedrocatalao, 2021-10-31:
It's hard to discharge 40% in 1 hour even with heavy processes running, it
seems clear that your mac is not sleeping and its battery health is not
perfect.

fragment75, 2021-11-01:
> It's hard to discharge 40% in 1 hour even with heavy processes running, it
> 
> seems clear that your mac is not sleeping and its battery health is not
> 
> perfect.

Thats what i thought, it started to behave like that when i installed al dente, but it doesnt bother me. I will do more testing when i remove the app

jirayu3141, 2021-12-28:
I have the same issue with my M1 Pro mac. Looks like it could be affecting mac's ability to deep sleep. As you can see, I did not use my computer at all on that day and it drains super quickly. 
[image]

ravenholt, 2022-01-13:
I'm on a Macbook 16" M1 Pro, and Al Dente seems to be causing massive battery drain on my machine during sleep as well. I'll try to do more testing later to confirm, but if I keep Al Dente installed I receive tons of these DarkWake messages, over and over, draining at a rate of 1%~ per hour.

`2022-01-13 13:25:47 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:25:53 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:25:59 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:05 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:11 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:17 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:23 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:29 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:35 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:41 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 5 secs
2022-01-13 13:26:46 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 6 secs
2022-01-13 13:26:52 -0500 DarkWake            	DarkWake from Deep Idle [CDN] : due to NUB.SPMISw3IRQ nub-spmi0.0x02 rtc/ Using BATT (Charge:81%) 6 secs`

` pmset -g
System-wide power settings:
 SleepDisabled		0
Currently in use:
 standby              1
 Sleep On Power Button 1
 SleepServices        0
 hibernatefile        /var/vm/sleepimage
 powernap             0
 disksleep            10
 sleep                1 (sleep prevented by powerd)
 hibernatemode        3
 ttyskeepawake        0
 displaysleep         5
 tcpkeepalive         0
 lowpowermode         0`

SergioSuarezDev, 2023-01-08:
i have the same issue on macbook pro intel 2017

Jasonzzb, 2023-06-21:
I got same problem with the newest version 1.22 on my 2020 intel Macbook Pro. before I use Aldente Pro, I put my mac into sleep, there were barly 1% battery lost over a night. but after I use it, there are 20% or even more battery loss.

alofting, 2024-02-23:
[image]

I experienced the same problem with my MacBook Pro M1 2021, Sonoma 14.3.1;  ~30pp drain per 24h in sleep mode until the battery was completely drained the second day sleeping with no screen usage. 

AlDente enabled options: 'Stop charging when sleeping', 'Sailing Mode' (5%), 'Heat Protection', 'Updates', 'Launch at Login'