AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #92: Better icon with status

pedrocatalao, 2021-02-01:
- Icon design more consistent with the rest of the system icons
- Indication if battery charging is limited or not
- Simple way to enable/disable the limit (e.g. right click)

Related: https://github.com/davidwernhart/AlDente/issues/74

nicho1asdev, 2021-02-02:
You can see if the charging is limited on the stock battery icon in the status bar. If it has a cable, it's limited, and if it has a thunderbolt, it's charging.

pedrocatalao, 2021-02-02:
@nicho1asdev that's true but if this story is accepted and picked it makes sense to adjust the icon to also show the app status without having to look at another icon.

TheMazus, 2021-02-14:
Hi pedrocatalao, your ideas sound great. I will definitely put it on our list to implement them.

CharlieBrown12, 2021-02-18:
Program is absolutely great.

1. One more ask for a nicer status bar icon...
How about replicating the Mac battery icon? With the status charging or not.
In that case "original" battery icon can be disabled in System Preferences.

2. Disable Get Pro button until its really implemented.
I would donate to remove it (which I tried today) just to see that its "fake".

Many, many thanks for this app..

stefandesu, 2021-03-21:
> 1. How about simply _replicating the Mac battery icon_?
>    With the status charging or not.
>    In that case "original" battery icon can be disabled in System Preferences.

I love this idea, would really appreciate this!

vassudanagunta, 2021-03-21:
Also it doesn't look like a plate of pasta at that size unless you look really close. More like a donut with a stick hovering over it. Or a plate with a streak on it.

I'd recommend a different name anyway, if you're hoping to get popular. A name that either evokes what it does (like [Caffeine](https://caffeine.en.softonic.com/mac) and [Amphetamine](https://apps.apple.com/us/app/amphetamine/id937984704?mt=12))  or that is catchy on its own. Though that's how I like my pasta, the term doesn't appeal to me on its own :)

If you're really attached to the name, you still need something other than a plate of pasta.

IMHO :P

MatthiasKerbl, 2021-03-22:
Hi @vassudanagunta,
thanks for the suggestions we will think about them.

Matthias

Madhawa97, 2021-04-02:
[image]
making it to match with the colour of other icons would be great for the time being.

MatthiasKerbl, 2021-07-05:
> * Icon design more consistent with the rest of the system icons
> * Indication if battery charging is limited or not
> * Simple way to enable/disable the limit (e.g. right click)
> 
> Related: #74

Hi @pedrocatalao,
we implemented your suggestions in AlDente Pro.
You can check it out here: https://apphousekitchen.com/

I hope you like it.
Matthias

closed 2021-07-05 by MatthiasKerbl