AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #106: Update brew formula

flying-sausages, 2021-02-14:
Currently 1.2 is released in brew. I'm not sure whether this is automated or not but it would be nice to change this to 2.0 so we can keep things up to date through this avenue.

https://github.com/Homebrew/homebrew-cask/blob/HEAD/Casks/aldente.rb#L2

vfl97, 2021-02-15:
Wish the app has an auto-updater too

lpla, 2021-02-15:
Given the new format in the release tag and .zip naming, updating it in brew had to be done manually: https://github.com/Homebrew/homebrew-cask/pull/99908

lpla, 2021-02-18:
Given that v2.0 and v2.1 are tagged as alpha and there are still issues regarding M1 support, I would say AlDente cask won't be updated until a stable version hits or latest alpha is considered stable enough by the users.

closed 2021-07-05 by MatthiasKerbl