AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #109: 2.0 alpha M1 - battery still sometimes charges

hgo11, 2021-02-15:
Hello,
First of all thanks a lot for all the efforts put into this, excellent it is working again.

I tested immediately after release and observed the following:
- started at 77%, charge inhibit kicked in immediately
- sporadically jumped into charging mode and by next day morning it was at 81%

It seems like sometimes the MBP modifies something and tries charging - until this is detected by AlDente some charge leaks. 

My setup: MBP, Apple charger that came with the book + a thunderbolt dock inbetween the charger and laptop.

vicemice, 2021-02-15:
I have the same issue on M1 Air. After studying syslog I found a new message:

```
systemstats[338]: assertion failed: 20D74: systemstats + 399948 [146262DC-951A-39E6-88F2-E8A328A3A263]: 0x0
systemstats[338]: assertion failed: 20D74: systemstats + 602280 [146262DC-951A-39E6-88F2-E8A328A3A263]: 0x2
```

Could you please elaborate how charge limit has been achieved since there's no SMC on M1 macs? Is possible to do a full rollback?

AppleRobert, 2021-02-16:
I have noticed the same behavior on my M1 MBA. I have the app set at 75% and the system battery reading was 80% which is what I wanted. The battery stayed at 80-81% until I either changed users or rebooted. When I do either of those things, the battery appears to charge. My battery percentage has risen to 85%. Last night the app did th same thing. I reinstalled several times making sure to reset battery to 100% and then doing a cold boot.

itsDaniel-23, 2021-02-16:
I experienced the same on my MBA. Had it charged to 50% yesterday and then dropped Aldente to 40% max. I didn’t shut down my Mac just closed the lid, left it on the charger all night and this morning my charge went up to 65%. Oh and prevent sleep is turned on even though in Aldente it says false.
[image]

DevinKott, 2021-02-16:
Same problem here. Yesterday, I couldn't get it to work. I installed while at 90%, and it was not working (still charging). Rebooted a couple of times and same problem. Plugged it in again today at 85%, and I saw that it did stop the charging for a couple of minutes but now it is back to charging.

Charge max has been set to 75%, so I'm way above the limit I set.

timension, 2021-02-16:
I can confirm that the charge is slowly increasing. About 1-2 percents per hour. I have M1 MBA.

davidwernhart, 2021-02-16:
Hey guys! Sorry to hear about your inconveniences. Sadly so far, I was not able to reproduce this behaviour on my M1 Macbook Pro. I have now compiled a new version that modifies the CH0C key instead of CH0B, which seems to be more conistent in it's behaviour. Also I fixed a bug regarding the disable sleep assertions. I attached this version for you to test if you want to, maybe it fixes the problems you are experiencing.

@vicemice , there actually is a SMC on M1 macs, allthough it is built into the M1 SOC instead of being a seperate chip. Since the keys for the new SMC are different, charging now has to be manually stopped by an active programm, for this reason, it does not work while rebooting or changing users.

Hope that clears some questions up.
Lots of love,
David
[AlDente_v20.01.app.zip](https://github.com/davidwernhart/AlDente/files/5991321/AlDente_v20.01.app.zip)

mkln, 2021-02-17:
The battery was slowly charging in my MBP M1 too. It went up 4% in an afternoon while also saying "not charging".
The behavior seemingly stopped once I disabled the MacOS "optimized battery charging" feature. No change in battery % overnight.

vicemice, 2021-02-17:
@davidwernhart Thanks! Is it possible to reset integrated SMC to defaults?

patryk-jason, 2021-02-17:
I'm very thankful to everyone who made it work on M1.
I can also confirm that it is still slightly charging Macbook when AlDente is turned on.
[image]

martinkasa, 2021-02-17:
I disabled "optimized battery charging" as @mkln said with limit to 70%. Currently, after 5 hours it is still stable on 70%.

davidwernhart, 2021-02-17:
@vicemice to reset the SMC on M1, you simply have to power down the notebook and close the lid for 30 seconds.

@patryk-jason have you tried the new version and disabled optimized charging?
btw, changes from the test version are new also available in the [newest](https://github.com/davidwernhart/AlDente/releases) release, plus automatic battery discharging.

patryk-jason, 2021-02-17:
@davidwernhart I'll try new version tommorow. Optimized charging was enabled. We'll see tommorow. Thanks

timension, 2021-02-17:
> @vicemice to reset the SMC on M1, you simply have to power down the notebook and close the lid for 30 seconds.

Do I understand it correctly that if I want to restore the Mac to the original state I just remove the app, switch off the computer and close the lid?

LeshawnG, 2021-02-17:
Confirming charging slowly occurs with optimized charging enabled. Not sure about disabled yet.
About to uninstall AlDente 2.0 (set to 100%first) and the helper as well from `/Library/PrivilegedHelperTools/com.davidwernhart.Helper` then power down and closing the lid for 30 sec. I'll turn off optimized charging before installing 2.1 then update again in a few hours after keeping the charger connected. 

Quick update on my setup and settings:
Using an 18W Apple charger with an MBP M1 13" (Big Sur 11.1) with AlDente 2.0 and optimized battery charging on.
Charging rate about 0.1-0.2%/hr (monitoring exact % using Better Battery 2 or Coconut Battery). In 24 hours, the macOS battery reading increased from 50-59%. Charger connected icon is always displayed.
The laptop was asleep, plugged in for about 3.5/4 hours (between 4-8pm) and it charged roughly 5% and the remaining 4% charged while I was actively using it later in the night.
[image]

AppleRobert, 2021-02-17:
I installed the build attached to the thread.  I have not experienced any battery charging overnight. I do not put my M1 MBA to sleep. I just turn the screen off.  The only time I see a increase in charging now, is  during reboot or changing users, which is to be expected.

patryk-jason, 2021-02-18:
With new version and optimized charging disabled charging stops exactly on the level I've set (MBP M1) I'm very happy with new discharging functionality - I was wondering about it several times. Very useful!

mkln, 2021-02-18:
> I think what is causing this issue is the "optimize battery charging" option in system preferences (as other have mentioned above). I've noticed the same behavior before unchecking that and now everything appears to be working properly. Battery percentage has not changed while plugged in for over 24 hours.

Yep, this is what I see after disabling optimized charging from system preferences: 

[image]

DevinKott, 2021-02-18:
When turning off the Apple Battery Optimization, the application works correctly and inhibits charging at the specified limit (which is great!). However, I have the limit set to 59% and it has slowly crept up to 62% within the last couple of hours. Will keep an eye on it.

When clicking the battery icon, it says "Battery Is Not Charging" when plugged in.

timension, 2021-02-18:
Can confirm that the latest version of the software works as expected even with the Apple Battery Optimization TURNED ON. No creep charge during last four hours.

chuanhhoang, 2021-02-19:
> Hey guys! Sorry to hear about your inconveniences. Sadly so far, I was not able to reproduce this behaviour on my M1 Macbook Pro. I have now compiled a new version that modifies the CH0C key instead [...]

Is the 2.1 compiled with CH0C or CH0B?

davidwernhart, 2021-02-19:
@chuanhhoang it is compiled with CH0C for M1 and CH0B for Intel (Since CH0C does not exist for intel)

Limeybastid, 2021-02-21:
Using version prior to 2.1 was solid, just deleted current version and installed 2.1
[image]

hgo11, 2021-02-21:
Thanks a lot, appears to be lock solid now.

closed 2021-02-21 by hgo11

timension, 2021-02-24:
Well, the case is closed but the issue with creep charge has unexpectedly come back. After it worked as it should for a week with constant SoC of 40%, during the last 24 hours it increased by 2% when the lid was closed. It's not a disaster and I can definitely live with it but still something is a little off.

I run AlDente 2.1 Alpha on MBA, M1.