AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #112: Macbook Air M1 can't set Max Battery Charge

noahjansen, 2021-02-17:
When trying to modify the 'Max Battery Charge' it always jumps back to 3%. Tried rebooting, reinstalling helper, restarting app, and checking/unchecking 'Use MacOS battery scale'. This is on Big Sur 11.2.1.

noahjansen, 2021-02-17:
^This was on v2.1 btw

davidwernhart, 2021-02-17:
Hey @noahjansen 
The "use macos battery scale" option is not available on the new 2.1 version. Are you sure you are using the right one?

basedrhys, 2021-02-17:
I had a similar issue when I installed via homebrew - I uninstalled the homebrew version and manually downloaded the latest version from the [Releases](https://github.com/davidwernhart/AlDente/releases) page. Worked great.

noahjansen, 2021-02-18:
You're both right, I was using v1.2 and misread it as v2.1 as I assumed homebrew would install the latest version. With v2.1 I can now change the slider. Thanks!

closed 2021-02-18 by noahjansen