AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #114: "Allow discharge" deactivates Thunderbolt display when in clamshell mode

marcoduering, 2021-02-18:
Hi!

I know this feature is in development, just providing feedback:

When I enable "Allow discharge (Beta)" while in clamshell mode with external display (via Thunderbolt, also charging via the display), the external display goes dark. This does not happen when using both the internal and the external display simultaneously.

Macbook Air M1
macOS 11.2.1
AlDente 2.1 alpha
LG Ultrafine 5K, connected via Thunderbolt
Charging via LG 5K

AppleRobert, 2021-02-19:
how do you set the "allow discharge" level? I don't see the option anywhere.

paoloestar, 2021-02-19:
same here with comparable setup

davidwernhart, 2021-02-19:
Hey guys!
I am aware of this issue. Sadly, as of now there is nothing I can do about it, since this seems to be a hardware limitation when telling the system to soft-disconnect the charger.
I have already mentioned this on the release-page for 2.1 alpha, but I should probably display this information somewhere more prominently.

Best regards,
David

marcoduering, 2021-02-19:
Thanks a lot for your feedback. Shame on me for not reading the release notes! Sorry! Closing this issue

closed 2021-02-19 by marcoduering

jamesdbailey, 2021-02-21:
When in clamshell you should try: sudo pmset disablesleep 1

aktivde, 2021-03-08:
Just came across an app called amphetamine that prevents MacBooks from falling into sleep mode when closed. Integrating such functionality into aldente could solve the issue?