AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #115: How Do You Set the Battery Discharge Percentage beta in 2.1

AppleRobert, 2021-02-19:
How do you set the battery percentage for the "Allow Discharge" beta option? All I see the tick box to enable the beta feature.

LeshawnG, 2021-02-19:
I believe this setting applies when the battery percentage is above the set level, it will discharge to the set level. So if you have 70% and AlDente set at 50% with "Allow discharge" enabled, it will discharge and stay at 50%.

AppleRobert, 2021-02-19:
> I believe this setting applies when the battery percentage is above the set level, it will discharge to the set level. So if you have 70% and AlDente set at 50% with "Allow discharge" enabled, it will discharge and stay at 50%.

Thank you for taking the time to reply, Leshawn.

closed 2021-02-19 by davidwernhart