AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #116: Option to disable charging when battery is above a certain tempurature?

vassudanagunta, 2021-02-19:
I would think this would be greatly beneficial for battery health.

TheMazus, 2021-02-19:
Hi vassudanagunta,
that's a great idea. We will definitely look into this.

CharlieBrown12, 2021-03-11:
Just a personal comment on how this can already be managed.

When AlDente is ON and you setup certain treshold - lets say 50% - once that charging threshold is reached - 2 things happen:
- charger delivers power directly to the system
- battery (not being charged) starts to cool-down

So if you have situations where you need to stress the system in longer periods while on charger - best is to start the day by setting AlDente % to the current battery charge %. This way - there will be no charging.

If you start the work with lets say 30% battery - and AlDente is set to treshold of 50%. Battery will heat during this charging gap of 20% in between.

As I said - immediate solution: lower AlDente to 30% to equal the battery level - and system will get the power from charger directly without overheating the battery.

MatthiasKerbl, 2021-07-05:
> I would think this would be greatly beneficial for battery health.

Hi @vassudanagunta,

With the introduction of AlDente Pro a couple of months ago, we introduced the feature Heat Protection which does exactly do what you suggested.

Thanks again and I hope you like it!

closed 2021-07-05 by MatthiasKerbl