AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #122: Checking Beta 2.1 on my MBP Intel - check hot wo discharge

tosbsas, 2021-02-20:
when Allow to discharge I can enable smc key, but, enabling that allow to discharge disappears - what should I do?

TheMazus, 2021-02-20:
Hi tosbsas,
you can uncheck "Use classic SMC Key (Intel)" and still use AlDente with the feature "allow discharge". Keep in mind, when "Use classic SMC Key (Intel)" is unchecked, AlDente will only not work when your MacBook is turned off.

tosbsas, 2021-02-20:
Ok. Are you saying that with the smc key disabled battery will charge when my mbp is off or on standby? 

What is the recommended setting for a non M1 devices ?

aktivde, 2021-02-24:
I also don't really understand the new checkbox "Use classic SMC Key (Intel)". You may add an explanation to the readme?

closed 2021-07-05 by MatthiasKerbl