AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #125: Charge limiting doesn't work unless application is running.

Averiun, 2021-02-20:
I don't know if this is the normal behaviour, but charge limiting doesn't work unless the application is running. 
My only problem with this is that the menu bar icon doesnt really look good nor does it fit with the stock macOS icons.

jacksonchui, 2021-02-20:
You should use something like [Dozer](https://github.com/Mortennn/Dozer) to hide it then.

TheMazus, 2021-02-20:
Hi Averiun,
are you on an Intel or Apple Silicon Macbook?

Averiun, 2021-02-20:
Hi!
I have the M1 MacBook Pro

TheMazus, 2021-02-20:
> Hi!
> I have the M1 MacBook Pro

For M1 MacBooks it is normal that AlDente has to keep running to limit charging. For now, you have to use other tools to hide the AlDente icon.

Averiun, 2021-02-20:
> > Hi!
> > I have the M1 MacBook Pro
> 
> For M1 MacBooks it is normal that AlDente has to keep running to limit charging. For now, you have to use other tools to hide the AlDente icon.

Alrighty, thank you!

closed 2021-02-20 by Averiun