AppHouseKitchen/AlDente-Battery_Care_and_Monitoring #129: Can someone explain this option

lixiao189, 2021-02-21:
I'm a mac book pro 2020 (intel) user, 

[image] 

So Do I need to click this option? Can some explain the function of it?

TheMazus, 2021-02-21:
Hi lixiao189,
When you are on an Intel MacBook, you can choose between "normal" mode (which is activated by default) and "Use Classic SMC Key (Intel)" mode. When "Use Classic SMC Key (Intel)" mode is activated, you will lose the feature "Allow Discharge (Beta)". However, you gain the feature that AlDente will keep your battery at the desired percentage, even if your MacBook is turned off. On the other hand, when "Use Classic SMC Key (Intel)" mode is deactivated, you gain the feature "Allow Discharge (Beta)" but AlDente limits your charging only while your MacBook is running or on standby.

lixiao189, 2021-02-21:
OK！Thank You, Sir!

closed 2021-02-21 by lixiao189